prusa3d/Prusa-Firmware #132: Question: Why is pin 84 defined twice in 'pins.h'

3d-gussner, 2017-06-26:
This is a minor question/issue, but still what to know why you defined the pin 84 twice in the 'pins.h' file?

**#ifdef SNMM** 

**#define E_MUX0_PIN 17** This pin is connected to the Prusa Multi Material Super Switch
**#define E_MUX1_PIN 16** Also this pin is connected to the Prusa Multi Material Super Switch
**#define E_MUX2_PIN 84** But this one is NOT connected to the Prusa Multi Material Super Switch which just uses a 4-pin connector VCC,PIN17,PIN16,GND to miniRAMbo P3. So why is it defined?

and later in the 'pins.h' you find

**#define BEEPER 84** Which is connected to the LCD board beeper.

Checking the firmware code i found **E_MUX_2_PIN** in 'ultralcd.cpp' and 'Marlin_main.cpp' where it stays always 'LOW'. The 4 extruders on the Prusa MM Super Switch are just controlled by the E_MUX0_PIN (17) and E_MUX1_PIN (16).
Doesn't it make sense to remove the 'E_MUX2_PIN' from the firmware when it is useless or may even interfere with the beeper?

PavelSindler, 2017-06-27:
Hello, 
yes it is redundant in SNMM section. I will remove it. Thank you very much.

3d-gussner, 2017-07-02:
Thanks, close the question

closed 2017-07-02 by 3d-gussner