prusa3d/Prusa-Firmware #134: Extra code - no function

thess, 2017-07-01:
While trying to clean-up the many dozens of compiler warnings in the code-base, the following code segment (**planner.cpp, line 961**) was flagged as having no effect. In particular the left-shift expressions are optimized out due to having no left-context.
```
#if 1
  // Oversample diagonal movements by a power of 2 up to 8x
  // to achieve more accurate diagonal movements.
  uint8_t bresenham_oversample = 1;
  for (uint8_t i = 0; i < 3; ++ i) {
    if (block->nominal_rate >= 5000) // 5kHz
      break;
    block->nominal_rate << 1;
    bresenham_oversample << 1;
    block->step_event_count << 1;
  }
  if (bresenham_oversample > 1)
    // Lower the acceleration steps/sec^2 to account for the oversampling.
    block->acceleration_st = (block->acceleration_st + (bresenham_oversample >> 1)) / bresenham_oversample;
#endif
```
You will notice that the `bresenham_oversample` value remains at 1 after exiting the loop. Fixing the left-shift operators to be `<<=` does have an effect on movement, but it seems to slow it down considerably. What was the intent of this code or should it be just `#if 0` or removed entirely?

bubnikv, 2017-07-05:
Thanks for pointing this out. That is an obvious bug. We will look into that.

bubnikv, 2017-07-05:
Funny thing, your pair of eyes were not the 2nd to read this piece of code.

Sebastianv650, 2017-07-05:
I was reading about Oversampling once ago in the Marlin discussion. Does it realy has an effect on print quality? I would think the real stepper resolution is much lower than any calculation or bresenham acuracy we will ever reach?

thess, 2017-10-07:
@Sebastianv650 - I think I agree with your analysis. When I fixed this code to actually do something, there seemed to be less than desirable side-effects.

bubnikv, 2017-10-08:
> When I fixed this code to actually do something, there seemed to be less than desirable side-effects.
@thess You mean the stuttering of the planner due to the increased load of the processor?

thess, 2017-10-08:
I don't recall if it was stuttering or just a general slow down of movement. Also not sure if there was any improvement of print quality vs. LIN_ADVANCE which was significant.

bubnikv, 2017-10-08:
The oversampling supposed to improve some alias effects visible on the diagonal straight lines. Actually I believed it worked, but it was just a placebo effect, as you found out.  I still believe the oversampling may be useful, but it may take more effort than I hoped to make it plausible on the Arduino board.

Sure the linear advance makes a much bigger difference at higher print speeds, but the oversampling is really orthogonal (read independent) from the linear advance.

thess, 2017-10-08:
I'm willing to give it another try with a bit more study. My main goal at the time was just to get a clean code-base with respect to build warnings and not analyse the intent of the code. There are times when I feel the firmware functionality (mathematically) is far more precise/critical than the hardware capabilities.

Given the discussion and my own local experience with changing the code as in #147 , I find that this mod (at least with my somewhat well aligned XY platform) gives nice straight lines without jogs in the Y axis.

Maybe @Sebastianv650 would like to try it too and give some additional feedback.

thess, 2017-10-10:
Right - I tried printing a test square (75x75) with both the oversampling enabled and disabled. AFAICT, the version with oversampling enabled runs slower and perhaps stutters as you have suggested. The runtimes for printing my sample were:
With oversample: 12:43
W/O oversample: 11:58

The difference is probably greater than it appears due to the warm-up time and mesh bed probing done before each print. The mesh bed probing even sounded different - FWIW.

Do you have a good sample which can demonstrate the difference??

bubnikv, 2017-10-11:
> Do you have a good sample which can demonstrate the difference??

Seemingly it never worked, shame on me, and due to the placebo effect I saw some differences. 

We would have to spend some time on it to make it working, but it may not be feasible with the current CPU.

To answer your question: As of now, we don't have an example to demonstrate the benefits of oversampling.

michalxfanta, 2019-04-11:
This issue is outdated.

closed 2019-04-11 by michalxfanta