prusa3d/Prusa-Firmware #139: 3.10.12-RC2 - Single Material - Automatically unloads filament after each print

MartinMajewski, 2017-07-04:
I have just tested the RC2 firmware, coming from the 3.10.11 stable release on my single material MK2 printer.

After each print the printer behaves as the MultiMaterial RC1 version did: it performs additional cleaning movements right after unloading the filament at each finished print. This is an annoying behavior.

Best wishes,
Martin

PavelSindler, 2017-07-04:
Hello @MartinMajewski , can you please provide gcode file where this issue appeared?

MartinMajewski, 2017-07-04:
I've tested it with e.g. the Z calibration for ABS file.

PavelSindler, 2017-07-05:
I need gcode file to debug what is wrong. It can be something in gcode itself. You should be able to add it here to reply.

closed 2018-12-11 by inquisitiveKangaroo