prusa3d/Prusa-Firmware #141: MM - Loading process loads filament too deep into multiplexing unit

MartinMajewski, 2017-07-04:
I have found an issue with the multi-material upgrade regarding the filament loading process.

It seems that the fast loading step (2) pushes the filament way too deep into the multiplexer. Therefore, loading another material into a second extruder unit causes both filaments so crash against each other. This crashing causes the drive gear to grind into the filament, which in turn diminishes the functionality of the gear because of debris between the teeth. Besides being a pain to clean the gear inside the small cabinet, the filament remaining cause problems along the way during print.

Furthermore, if the first extruder that is loaded is not the same that is used first on the next print job, there is no chance the actual material can be extruded because the wrong material is still inside the multiplexer and even the cold end.

Here is a picture showing how deep the fast loading step pushes the filament inside. I've pressed the button as soon as the hob caught the material.

[image]

If I am doing something wrong, please instruct me on how to use the mm-upgrade properly.

Firmware used is 3.0.12-RC2

Best wishes,
Martin

PavelSindler, 2017-07-04:
Press reset x button and then during reset hold knob. Factory reset menu will appear. Scroll to last item in menu which is "Bowden length" and confirm with pressing the knob. Check if bowden length is set to 385 mm. If it isn't write me here what the length in menu is and set it to 385 mm. If it will not help, check what physical bowden length is. I am sorry for inconvenience.

MartinMajewski, 2017-07-05:
As seen in the pictures below, everything is already set to 385. The physical bowden length is also 385 mm.

[image]

[image]

I've repeated the unloading / loading procedure several times now. Each time the same result. The filament is pushed between 85 mm to 90 mm further out of the bowden tube.

[image]

MartinMajewski, 2017-07-05:
Setting the Bowden length value to 335 mm produces much more reasonable results.

GurliGebis, 2017-07-05:
Might it be a problem with the number of E steps? (like it running too many steps for each milimeter it tries to push out)

Martin, could you try the first part of this guide, to see if the number of e-steps are wrong? https://mattshub.com/2017/04/19/extruder-calibration/

GurliGebis, 2017-07-05:
(Just do it with only one filament loaded)

MartinMajewski, 2017-07-05:
Oh yeah, the E steps are off.

Original settings:
E161.30

This value resulted in an extrusion of 115.00 mm.

Doing the math for 115.00 mm = 161.30 stp over 100.00 mm = x stp results in 

`100.00 mm * 161.30 stp / 115.00 mm = 140,26 stp`

I entered this number and got a perfect stop at 100 mm filament length.
So I did the reset of the bowden length value back to 385 mm and got a filament overshoot of 25 mm (compared to the nearly 90 mm previously).

Does this sound legit?

PavelSindler, 2017-07-05:
Yes, it sounds good.
Use `M92 E140` to set steps per unit to correct value for multimaterial. Then store settings with `M500`. Finaly set bowden lengths back to 385 mm and it should be ok.
It seems that we had bug there wich causes that users who upgrade from singlematerial and used `M500` for storing custom settings (like PID cal. results) can have wrong Esteps now. 
Thank you very much for you feedback.

closed 2017-07-17 by PavelSindler