prusa3d/Prusa-Firmware #146: problem temperature fw 3.0.12

shikyoo, 2017-07-13:
hi
when i install the firmware 3.0.12 on my mk2 (not multi material) the temperature cannot up to the temp define for my print.
example:
when i print a pla with 210°C the temperature stay in 200°C
when i print a petg with 250°C the temperature stay in 240°C

i restore my firmware on 3.0.10 and i don't have problem with temperature

JohnOCFII, 2017-07-13:
As a data point, with 3.0.12-RC2 I do not have this issue.  If I choose 250C, the hot-end will hold 250C.

GurliGebis, 2017-07-13:
@shikyoo Is the problem present in the 3.0.11 release?
And if not, could you try the two RC versions of 3.0.12 to see when the problem was introduced?

macfly1202, 2017-07-13:
I didn't have this issue on the last release 3.0.12 rc2. I had rc1 installed previously but I didn't look enough during print to see this. 
In parallel, I redo PID calibrationy temperature at each major release.

GurliGebis, 2017-07-14:
Does it work if you reset the stored temperature parameters?

shikyoo, 2017-07-24:
Sorry for my delay
I have the same problem when I installed firmware 3.0.11, I haven't tested the 3.0.12 rc2 I will try thanks for your help

djolmos, 2017-08-08:
I just installed the FIRMWARE 3.0.12 on my mk2 and im having the exact same problem. @shikyoo

Sebastianv650, 2017-08-08:
Has anyone of the users having problems actualy tested if doing a PID tune solves the problem? On my MK2 without any FW changes freshly built up out of the box the temperatures were also quite instable, a simple PID tune solved that.
I'm running 3.0.12 and my temperature is stable within +-2°C.

closed 2018-12-11 by inquisitiveKangaroo