prusa3d/Prusa-Firmware #147: Surface artifacts caused by XY skew compensation

ceryen, 2017-07-14:
On my printer, I was seeing surface artifacts on surfaces parallel to the X-axis or Y-axis:

[image]
[image]

My hypothesis was that these surface artifacts were microstepping artifacts caused by XY skew compensation.  Although the firmware reports that X-axis and Y-axis are perpendicular to each other, the firmware actually leaves skew compensation turned on in an attempt to compensate for a very small skew angle.  As a result, when printing surfaces that are parallel to the X-axis or Y-axis, individual microsteps are introduced to compensate for the very small skew angle.  Unfortunately, stepper motors cannot accurately insert individual microsteps, resulting in microstepping artifacts in the print surface.

To test this hypothesis, I modified world2machine_update() to ignore the passed in 'vec_x' and 'vec_y' vectors and to simply set the 'world2machine_rotation_and_skew' matrix to the identity matrix:

    world2machine_rotation_and_skew[0][0] = 1.f;
    world2machine_rotation_and_skew[1][0] = 0.f;
    world2machine_rotation_and_skew[0][1] = 0.f;
    world2machine_rotation_and_skew[1][1] = 1.f;

This then disabled XY skew compensation and I saw that the surface artifacts disappeared from my prints:

[image]

I would be interested in seeing whether others can reproduce these results.  If these results are reproducible, I think it would be an interesting discussion as to whether compensating for a very small skew angle is worth introducing microstepping artifacts in the print surface.

===================

Note that I actually tried to disable XY skew compensation by turning on the following code in calculate_machine_skew_and_offset_LS():

    #if 1
    if (result == BED_SKEW_OFFSET_DETECTION_PERFECT && fabs(a1) < bed_skew_angle_mild && fabs(a2) < bed_skew_angle_mild) {
        if (verbosity_level > 0)
            SERIAL_ECHOLNPGM("Very little skew detected. Disabling skew correction.");
        // Just disable the skew correction.
        vec_x[0] = MACHINE_AXIS_SCALE_X;
        vec_x[1] = 0.f;
        vec_y[0] = 0.f;
        vec_y[1] = MACHINE_AXIS_SCALE_Y;
    }

However, I found that this did not actually disable XY skew compensation.  That is because the 'vec_x' and 'vec_y' vectors are later modified by:

    // Invert the transformation matrix made of vec_x, vec_y and cntr.
    {
        float d = vec_x[0] * vec_y[1] - vec_x[1] * vec_y[0];
        float Ainv[2][2] = {
            { vec_y[1] / d, -vec_y[0] / d },
            { -vec_x[1] / d, vec_x[0] / d }
        };
        float cntrInv[2] = {
            -Ainv[0][0] * cntr[0] - Ainv[0][1] * cntr[1],
            -Ainv[1][0] * cntr[0] - Ainv[1][1] * cntr[1]
        };
        vec_x[0] = Ainv[0][0];
        vec_x[1] = Ainv[1][0];
        vec_y[0] = Ainv[0][1];
        vec_y[1] = Ainv[1][1];
        cntr[0] = cntrInv[0];
        cntr[1] = cntrInv[1];
    }

As a result of these floating point operations, 'vec_x' and 'vec_y' are no longer exactly equal to (1.f, 0.f) and (0.f, 1.f).  As such, when 'vec_x' and 'vec_y' are passed into world2machine_update(), world2machine_update() turns on XY skew compensation even though calculate_machine_skew_and_offset_LS() intended to turn off XY skew compensation.

To confirm this, I printed the values of 'vec_x' and 'vec_y' and saw that they were actually equal to (1.f, 0.f) and (-0.f, 1.f).  I also printed the value of 'world2machine_correction_mode' and saw that it was 3, indicating that XY skew compensation was turned on when it was intended to be turned off.

ceryen, 2017-07-14:
Here is the STL and config.ini that I used for two of the photos above:

[Thin wall calibration.zip](https://github.com/prusa3d/Prusa-Firmware/files/1148680/Thin.wall.calibration.zip)

PavelSindler, 2017-07-17:
Thank you for very informative description. We will look at this closer.

GurliGebis, 2017-07-17:
@ceryen would it be an option to have a variable that disables the correction, and then set it to true in your first code snippet, and then in your second code snippet read that variable, and handle it correctly there?
I don't know if the entire invert part of the code should be disabled, or just the part setting the two vectors, but both should be possible to do.

ceryen, 2017-07-17:
@GurliGebis I don't have any preferences on how the code might be changed, I leave it up to the Prusa team to determine what the right fix might be =P  It might be good to know first whether others can reproduce this issue

GurliGebis, 2017-07-17:
In your test, you changed it to 1, and commented out the last code section in your test?

ceryen, 2017-07-17:
@GurliGebis No, I said that I turned on the code by changing the "#if 0" to "#if 1", but that didn't end up turning off the XY skew compensation code due to the invert code.  As such, I ended up modifying world2machine_update() instead to set 'world2machine_rotation_and_skew' to the identity matrix.  But that's just a quick hack for testing, I don't have any preferences for how it is eventually fixed.

thess, 2017-10-04:
Couldn't resist adding an interesting footnote here -- @ceryen, you analysis is indeed correct however, merely changing the `#if 0` to `#if 1` in the above code is not sufficient to disable skew compensation if the bed alignment is BED_SKEW_OFFSET_DETECTION_PERFECT as you already observed.

Changing:
```
#if 0
if (result == BED_SKEW_OFFSET_DETECTION_PERFECT && fabs(a1) < bed_skew_angle_mild && fabs(a2) < bed_skew_angle_mild) {
```
to:
```
#if 1
if (result == BED_SKEW_OFFSET_DETECTION_PERFECT) {
```
makes it all work. The testing of angles a1 & a2 is already taken into account earlier in the routine.

closed 2019-01-29 by inquisitiveKangaroo