prusa3d/Prusa-Firmware #148: Move extruder to suitable position before loading filament

probonopd, 2017-07-17:
The "Load filament" command should move extruder to a suitable position before executing the command.

Currently, when the user selects "Load filament" while the extruder is in position 0 (close to the bed) on the X-Axis, it produces a mess. 

Instead, the extruder should first move to a "safe" location, and then load the filament.

GurliGebis, 2017-07-17:
The only problem I can see with this, is if you are in the middle of a print, and you are close to running out of filament.
Then you would normally pause the print, move the print head away, unload and then load the new filament.
In that case, it would be unsafe to try and move the print head automatically, since there is a chance of the print hitting the print head or the X carriage rods.

probonopd, 2017-07-17:
Maybe this should be done only if the position is close to the bed.

GurliGebis, 2017-07-17:
Or even better, if there is no paused print, that way we know that there should be nothing in the way (at least in theory).

I'm thinking something along the lines of doing the following:

1: Autohome
2: Move Z axis to 180 (Should be high enough to allow for cleaning the nozzle while we are at it).
3: Move X axis to 125 (Should be the middle, which should give plenty of room to work with).

What do you think?

probonopd, 2017-07-17:
Maybe better move X all the way to one of the sides, so that the filament can fall off the build plate?

GurliGebis, 2017-07-17:
That would also work, but wouldn't you normally clean the nozzle while you are changing the filament? (I normally do that, but I don't know if that is the norm).
I know cleaning it when it is not in the middle is more difficult than if it was in the middle.

I don't think it is a problem having it in the middle at a somewhat high altitude, since it will just fall down on the plate, and should not stick to it (since it will have had time to cool off)

Sebastianv650, 2017-07-18:
I would avoid to move the nozzle to a fixed position in any case. This is quite the same as the M600 pause / filament change command where you find a as far as I remeber long discussion and development on Marlin.
Also don't do an auto-home: If you do that during a print to change the filament, you may end up with shifted layers because the endstops will never trigger at the exact same position.

There are always cases where a movement will fail or damage the print and it's not always possible to catch this exceptions. Therefore, if you think about any move have in mind:
- The system might not be homed, autohome without a warning like "XYZ Homing will be performed! Continue?" is questionable.
- A print might be running, so every down movement is impossible. A movement upwards might be impossible on large prints.
- If we have a print on the print bed, the extruded material should never hit the unfinished print. Therefore, only minimum or maximum of XY are suitable. Y max is also not possible because the print might make the nozzle unreachable.

I guess there are a lot more things to consider I'm not aware of at the moment.

GurliGebis, 2017-07-18:
I see - it is not an easy problem to solve, when you have to be sure it works in all situations.

3d-gussner, 2020-07-13:
@probonopd If I am not wrong that has been fixed, can you please confirm and close this issue?
[stale][fixed]

probonopd, 2020-07-30:
Closing for now since I can't retest at the moment. Feel free to reopen if the issue persists. Thanks everyone.

closed 2020-07-30 by probonopd