prusa3d/Prusa-Firmware #149: Extend the menu timeout values for "Move axis" menus

bubnikv, 2017-07-18:
by @bardiir:
Where I most definitely noticed it being annoying is the move menus when manually moving any of the axis. Those should in my opinion have very long timeouts as you most likely use them during other maintenance operations.

And overall I think you should increase the timeout by about 50%, it just feels very quick to drop out of the menu again if you look something up in the manual for example while navigating the menu.

See
https://github.com/prusa3d/Original-Prusa-i3/issues/33
for the initial report.

ChiefPoints, 2017-08-20:
agreed, at least a 50% increase would be good, for just myself I'd probably increase it a minute or remove it completely.

domesticatedviking, 2017-08-23:
Agreed.  I also think this menu is buried deeper than it should be.  The move axis menu belongs in the top level menu IMHO.

TonyLHansen, 2017-08-23:
I've often thought that there needs to be a "quick access" top-level menu that offers a few simple commonly-used items that are otherwise buried rather deep. For example, "move Z up 5cm". (Yes I know that there is no buried menu item that says "move Z up 5cm", but that is something I find myself wanting to do quite often.) The items that are available on this "quick access" menu should be individually choosable.

inquisitiveKangaroo, 2018-12-17:
There is a shortcut imlpemented for Z-axis moves by pressing and holding the knob. On the other hand I understand those timeouts might be annoying expecialy during the maintanance or any sort of testing.

closed 2019-04-11 by michalxfanta