prusa3d/Prusa-Firmware #150: Printer skipping the second layer

fucx, 2017-07-18:
Hi there,

I’ve noticed some strange behaviour with my MK2S. I’ve sliced a 25mm cube with a single outer and bottom shell and no top shell with 0 infill to calibrate the extrusion multiplier.
While watching the print, I noticed that the Z height on the LCD goes from 0.2 to 0.4 (for about half a second) and then prints the second layer at 0.6 mm height. I have checked the gcode file with Slic3r, gcode.ws, gcodeanalyser.com and Simplify 3D, all of which show correct layer heights for the first, second and third layers.

Ondrej (via support email) asked me to upgrade my printer’s firmware to 3.0.12, which I did, but the issue still persists. I even used a different SD card than the one I normally use, but that didn’t make any difference either.
He also asked me to create a video, you can find it here: http://d.pr/v/TTMzue/1MIqCX1B 
The first 1:20 is just the first layer, the switch from 0.2 to 0.4 and then to 0.6 happens at 1:23.

Here the gcode file: [cube.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/1156271/cube.gcode.zip)

From what I can tell, it’s printing the first perimeter at 0.2 from line 325 to line 332, then infill until line 509. Second layer starts at line 511 at a Z height of 0.4, third layer is at line 532 at Z height 0.6. That doesn’t look out of place to me. I’d really appreciate your insight here. 
Please let me know if you need any additional information.
Thank you!

Sebastianv650, 2017-07-18:
You video is 270MB.. Maybe you can shrink it a bit? ;-)
The gcode looks OK. Are you sure it's realy not printing layer 2? I would guess it's only the display skipping the 0.4 - as far as I know it's not showing the actual nozzle position but the final position at the end of all buffered moves. As your layer has only 4 print moves, the end of the buffer will point to something in the 3rd layer quite fast.

fucx, 2017-07-18:
Hm, you may be right about the buffer being the reason for this. Looking at the cube under a microscope definitely shows all layers having the same size, but applying minimal pressure to the base layer rips it off the perimeters. Also, looking at the first couple of layers under a microscope definitely shows something's off (larger distance between first and second layer, second and third layer squished together): http://d.pr/i/Vn9Ozy/3mdlCJFA (bottom layer is at the top of this picture)

I've reduced the video's size and have re-uploaded it: http://d.pr/v/LoWHE0/1yZwABgQ
You should now also be able to play it from the above location.

Sebastianv650, 2017-07-19:
I'm sure it's due to the buffer, the video isn't looking as it would skip a layer and you picture isn't also. As the printer isn't knowing about 1st or other layer numbers, there is also no reason inside the code it should handle 0.2 other than 0.4 or 12mm.

The visualy different layer heights in your picture can have a lot of reasons, mainly:
- Maybe it's only a visual thing, due to lighting and visual resolution
- Not perfectly calibrated 1st layer height (The prusa manual with it's "adjust by watching during printing" isn't very helpful at this point.)
- Not perfectly calibrated extruder steps/mm value

If you are happy with your normal prints, I wouldn't change something.

fucx, 2017-07-19:
Thank you for your insight. Let me just add my 2 cents:

- if it was merely a visual thing, I wouldn't be able to flick the perimeters off the base layer. this is not the case for cubes I printed a while ago, the first layer would rather rip than let go of the perimeters.
- as I mentioned, the first layer is 0.21mm high, which is close enough, I guess. I haven't calibrated it through "watching" but measuring (and printing lots of 75mm squares) for getting it there
- also, extruder steps per mm and extrusion width are both calibrated, as you can see below

This is what 0.45mm extrusion width yields in terms of actual perimeter width:
[image] 
So I think the printer's calibration shouldn't be an issue here. 

I guess I'll just accept that this issue exists and move on, then. I'll keep an eye on these kinds of prints, maybe it'll get fixed as a side effect of a firmware or Slic3r upgrade in the future. "Normal" prints with infill or more perimeters should be fine, I suppose.

Sebastianv650, 2017-07-19:
In fact it seems I skipped the line where you mentioned the 0.21mm layer thickness.

I would recommend to disable z hops during retract and slow down the print speed a lot so it's easy to see what's going on. Also an M600 pause at the end of the second line will give you the ability to take a look and pictures of what happened.

I realy can't believe this is due to an Slic3r or FW error. If there are such basic errors, the issue page would be flooded with entries. Printing cubes is something that 99% of the users might have done a lot of times. Therefore I hope you can excuse I skeptical to this one ;)

Easy layer separation on a transition like here from 1st to second layer is not that unusual. On one side it's some kind of infinite cross section change, which means high notch stress if this translation is correct. With the default Prusa Slic3r settings it's also highly likely that the always high enabled fan will cool down the single perimeters much too fast, resulting in very low layer bonding.

fucx, 2017-07-19:
No apology needed, I've been doubting myself over this more than I'd like to admit, but I'm able to reproduce this reliably, which is why I created this issue.

I'll give the tips you mentioned a try and will see how it goes. Thanks for your help so far!

closed 2018-12-28 by inquisitiveKangaroo