prusa3d/Prusa-Firmware #152: Enable move axis to negative values

bardiir, 2017-07-20:
If I have the printhead somewhere in the mid of the Z axis I can't move it down manually after powering on the printer. I can only move it up with the menu. I think it should be possible to manually move it down until the PINDA probe triggers, or at least have an option to unlock this for advanced users. If I ram my nozzle into the printbed it's my own fault but having to power of the printer and move the printhead closer to zero manually is somewhat cumbersome.

Sebastianv650, 2017-07-20:
Moving beyond the software endstops is no good idea for most of the end users in my opinion. The right way to do it would be to home the axes, I just checked on my i3 and I see it's not possible on the very limited options it gives you.

Would a home axis menu also fulfill your needs? If yes, maybe the Prusa team might include this - at least this is more likely than they allow to move beyond the endstops I guess :-D

GurliGebis, 2017-07-20:
You can home it from the settings menu (or is it the calibration menu, can't remember) :-)

bardiir, 2017-07-20:
Yes you can home, but somtimes there is no space to home all the axis, you just want to move one. Or at least I had this case this morning when leveling the PINDA probe. I wanted to move the head down 1mm without loosing the x/y position because the PINDA was resting on a leveling gauge. But I could not move Z down as it was at 0mm position after powering up.

If you move beyond the endstops that should not be a problem, if the hardware endstops are there they should trigger anyway once the axis reached real zero positioning. You can run into the other end without a limit too. So I don't see how stopping at 0 does protect you much.

GurliGebis, 2017-07-20:
The problem is, that the endstop for the Z axis is the pinds detecting the probe point. So if the bed is not at the correct position, it will not find the hardware endstop, if you could go below 0. Homing is safe, because the bed is at the correct position before the Z axis go down

GurliGebis, 2017-07-20:
Pinda*

bardiir, 2017-07-20:
If the PINDA is not mounted correctly it will ram the head into the bed full power anyway. You can't really protect users from everything. But this is why I suggested this as an advanced users option for people who know the dangers of ramming the nozzle into the printbed, maybe even make it a temporary safety unlock, once you power cycle/reset the printer safety nets are back online.

Sebastianv650, 2017-07-20:
@GurliGebis you are right, it's in the calibration menu! Well, I would be intrested in the thoughts that lead to this location X-)

@bardiir 

> If you move beyond the endstops that should not be a problem, if the hardware endstops are there they should trigger anyway once the axis reached real zero positioning. You can run into the other end without a limit too. So I don't see how stopping at 0 does protect you much.

The hardware endstops are usualy only active during homing and not checked during normal moves. In Marlin there is a switch to control this behaviour in Configuration.h, but I can't find it in Prusa FW. I guess it's hardcoded.
Yes you can run into max limits, but that's not that bad. Zmin is the only one that will create real damage.

If you realy want to try it, may change your Configuration file:
#define min_software_endstops true

GurliGebis, 2017-07-20:
@bardiir Have you tried using the gcode (cannot remember which one it is) to set the current possition.
That would be a way to trick it to thinking it is not at the bottom, though it would have to get sent over USB.

Sebastianv650, 2017-07-20:
http://reprap.org/wiki/Mendel_User_Manual:_RepRapGCodes#G92:_Set_Position

bardiir, 2017-12-19:
@Sebastianv650 no, actually I don't want to use a customized firmware as a potential additional error source. So this option is out of question.
@GurliGebis I don't really want to send commands to the printer via software for this, it's not connected most of the time.

--

But overall I do understand the requirement or intention behind the software min endstops so I guess there is no real/ideal solution to this ticket, but thanks for the input everyone :)

closed 2017-12-19 by bardiir