prusa3d/Prusa-Firmware #154: Semantic versioning

lekoder, 2017-07-21:
Is there a specific reason Prusa-Firmware does not follow semantic versioning?

[Semver](http://semver.org/) is a widely adopted practice of numbering software versions with `major`.`minor`.`patch` numbers (ie. 3.0.12), *and* incrementing:

- `major` when introducing breaking API changes (stuff not working with this software anymore)
- `minor` when introducing new features
- `patch` when fixing bugs.

It seems that Prusa-Firmware increments patch section only, and fixes bugs with prerelease postfix (ie. 3.0.12-1). This does lead to some confusion.

Please consider adopting semantic versioning in this project.

probonopd, 2017-07-21:
Prusa Firmware is a set of patches against Marlin. So the versioning should follow Marlin versioning, with an appendix for the Prusa changes version.

lekoder, 2017-07-21:
@probonopd If you are refering to [MarlinFirmware/Marlin](https://github.com/MarlinFirmware/Marlin/releases), I fail to see how current 3.0.12 corresponds to 1.1.4.

probonopd, 2017-07-21:
Yes, that's what I mean - wouldn't 3.0.12-1.1.4 better?

Sebastianv650, 2017-07-21:
Is prusa fw really based on marlin 1.1? Hard to believe, the portions of code i changed for LIN_ADVANCE was not looking like a recent marlin version at all. . Maybe more 1.0.

lekoder, 2017-07-21:
@probonopd This actually depends; does this repo get merges from Marlin, or is it developed independently?

- If there are merges coming in, it would be good to have -1.1.4 postfix in release (to know on what it is based).
- If not, that info is irrelevant.

Either way, right now we will be presented with 3.0.12-1, which I believe to be a bugfix release. It looks awkward to me, yet I am not maintainer of this repo. Hence original suggestion.

bubnikv, 2017-07-21:
Our firmware versioning is
major.minor.revision-suffix
where suffix is alpha, beta, rc, rc1, rc2, up to rc9

Our firmware version has this fixed format, so the firmware may receive a
"newest" firmware version through a g-code to remind the user to update the
printer.

Our Marlin fork has been forked somewhere around February 2006. Our
versioning is independent of the current Marlin. This is understandable as
the two firmware sources diverged a lot.

GurliGebis, 2017-07-21:
@bubnikv Is it really that old (2006) ?

bubnikv, 2017-07-21:
I meant 2016 naturally :-)
I am old, so ten years is under my resolution threshold :-)

GurliGebis, 2017-07-21:
@bubnikv Hehe, that sounds familiar.
That got me thinking, since it is a year since you branched out from the Marlin source tree, what plan do you have to keep up with them, or has it diverged so much that pulling features from them is more difficult than re-implementing them yourself?

closed 2018-12-28 by inquisitiveKangaroo