prusa3d/Prusa-Firmware #158: Auto home should move z-axis first

RavensburgOP, 2017-07-29:
**Issue**
There is a risk of damaging the heat bed when using the auto home option, as it first moving the x, then y and finally z axis. To solve this the z axis should be moved before the other axes.

**Example**
I was calibrating the z-axis after moving my printer. I noticed that it was bending the heat bed and turned it off to adjust the PINDA probe. To improve my working position I wanted to move the print head and therefore used the auto home option, unaware that the x-axis would be the first to be used. Before I could react the nozzle was rammed into the heat bed leaving a several millimetre deep tear in it.

robrps, 2017-07-29:
The probe only works when it detects the metal in the specific 9 points on the heated bed, if you were to home Z without homing X and Y first, you would miss those points and the Z would carry on....heated bed and PEI really don't like that happening!

RavensburgOP, 2017-07-29:
I'm not talking about homing z-axis first. I'm talking about moving it 1 cm up before starting movement on the x-axis.

robrps, 2017-07-29:
Ah OK, I see.

I would put that down to user error, it sounds like you left the nozzle pushing the bed down after powering off? You can manually rotate the two lead screws with it turned off, but you can end up with one lead screw slightly higher. You would need to move Z axis fully up after its powered on (it will only let you move up).

It doesn't help you now, but you'll remember not to do it again.

--Captain Hindsight.

_Sent from my OnePlus ONEPLUS A5000 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

RavensburgOP, 2017-07-29:
It wasn't touching the bed when I turned it off, but because the bed wasn't level it hit it. My point is still very valid though since all of this could be avoided with two extra lines of code. There's no warning at the moment, and it literally means that one wrong push on a button can ruin your 70£ head bed.

ManuGithubSteam, 2017-08-04:
I also have seen the issue @RavensburgOP  does describe. It is an easy fix just to move Z first 1 cm up when auto home. This will lead to fewer print beds damaged and is a work who should be done in less than five minutes.

jspiro, 2017-08-04:
Me too. There is no scenario that comes to mind where this would cause a more serious problem than the one it solves.

fiveangle, 2017-08-04:
On the rare chance that it was left at max z at power off, carriage will be driven up into the z ends so it might make sense to set to silent mode before moving z up, then return to default before starting the home process:

```
if (SilentMode == 1) digipot_current(1, DIGIPOT_MOTOR_CURRENT); //set back to normal operation currents
		else digipot_current(1, DIGIPOT_MOTOR_CURRENT_LOUD);
```

closed 2018-12-11 by inquisitiveKangaroo

RavensburgOP, 2018-12-12:
@inquisitiveKangaroo Thanks for adding the labels to the issue, though I'm curious why the issue was closed with no apparent fix. Could you explain the reasons behind this? :)

inquisitiveKangaroo, 2018-12-12:
Hi Christian,
Thank you for the topic and I apology for not commenting the reasons before closing. This issue is fixed on mk3 fw and when we release the mk2/mk3 merged firmware it shall be working for mk2 too. The merge should be released anytime soon.
I closed the issue in order to create some kind of 'order' in this repository. I'd like to keep open only issues we are truely working on and the other (solved, irelevant, low priority) close to keep it easy to orientate. We can always reopen the issue and we will (like in case the merge would not work as expected).
We are grateful for feedbacks like yours and we do not want these to get lost in ether.
Best regards, Kangaroo