prusa3d/Prusa-Firmware #159: Too Much Yet Too Little

CNealWJr, 2017-07-31:
Can the machine be set up to allow use of the MM to use the four filaments in series.  If you have a number of short rolls that need to be used up one could put them on and leave the machine for extended prints.  When one roll was empty the machine would move to the next and so on.  It would allow for the use of rolls that have too much filament left to toss out but not enough to allow for an extended and unattended print.

Thanks

Neal

fucx, 2017-07-31:
The problem I'm seeing with this is how the machine is supposed to know when the filament on one spool runs out, and hence when to switch to the next one.

CNealWJr, 2017-07-31:
Filament sensors?  I'm kinda ignorant about this but couldn't sensors be used on each feed motor and when it senses no more filament the machine would move to the next spool and so on down the line.

fucx, 2017-07-31:
Sure, a runout sensor could be used for this. However, this is the stock firmware, and the stock printer doesn't have a sensor. So I don't see this feature making it into the firmware unless the printer gets runout sensors.

ManuGithubSteam, 2017-08-04:
It is also possible to count the meters on the spool.
If the printer would assume a full spool on all 4 motors it can count the meters. So it knows when the filament is empty. It just needs to save these values somewhere so it knows where it left of.

m-Th, 2017-08-15:
Filament Sensor: a badly needed upgrade. After this, we will talk about firmware.

inquisitiveKangaroo, 2018-12-28:
Added on MMU 2.0 as the SpoolJoin feature (also called autoDeplete)

closed 2018-12-28 by inquisitiveKangaroo

inquisitiveKangaroo, 2018-12-28:
#167