prusa3d/Prusa-Firmware #161: Cura support 3.0.12

Thijxx, 2017-07-31:
As of version 3.0.12, the support for Cura is worse than it was. 

- It is not possible to cancel a print.
- It is not possible to view progress of the print.
- The printer does not show print time anymore.

As far as I know, Cura was the only software working with the Pruse i3 MK2s, without moving SD cards around.

Cura: 2.6.0
Pruse FW: 3.0.12

Sebastianv650, 2017-07-31:
> It is not possible to cancel a print.

You mean canceling by hitting calcel in Cura? That's not related to the printer, canceling a print only means stopping sending new gcode to the printer. There is no G or M code to cancel a print.

> It is not possible to view progress of the print.

On printer LCD or inside Cura? While the printer doesn't know how many lines of gcode will come until the end, it can't show the progress when printing over USB. If it's not shown inside Cura, it's a Cura problem.

> As far as I know, Cura was the only software working with the Pruse i3 MK2s, without moving SD cards around.

I'm using Pronterface for printing on both of my printers without problems. Also Slic3r is able to print directly over USB since some time.

Thijxx, 2017-08-01:
Aborting the print was possible before. Hit abort, the printer stops. Now there is no more Abort button, since I did not change Cura but did update the firmware, I think that is causing it.

Before, the printer timed the print, so the display showed minutes, then Cura showed the % sent to the printer, this way I could figure out when the printer would finish. With the latest firmware, the timer does not start.

Pronterface is new to me, I will check it out, thanks for the tip! Slic3r tries to destroy my printer, I tried the Prusa Beta many times and it turn my printer completely haywire so I gave up on it.

fiveangle, 2017-08-01:
Slic3r Prusa Ed. is working well for tens of thousands of MK2 owners (and who knows how many non-MK2 owners) so your report that it "destroys your printer" and "turns [your] printer copletely haywire" is difficult to take seriusly

Sebastianv650, 2017-08-01:
> Aborting the print was possible before. Hit abort (in Cura), the printer stops. Now there is no more Abort button, since I did not change Cura but did update the firmware, I think that is causing it.

The printer is only receiving a gcode stream over serial (USB). It can't control the visibility of a cacel button, so you in the wrong branch here.

> Before, the printer timed the print, so the display showed minutes, then Cura showed the % sent to the printer, this way I could figure out when the printer would finish. With the latest firmware, the timer does not start.

See if the M codes helps you: http://marlinfw.org/docs/gcode/M031.html

Thijxx, 2017-08-01:
I've installed OctoPrint, the progress timer and cancel work great! I also made gcode using the 'thousands of people say it's ok so it must be ok for you too' Prusa Slic3r and the result is good, so I will stick to this config for now :) thanks y'all

closed 2017-08-01 by Thijxx

Sebastianv650, 2017-08-06:
> The printer does not show print time anymore.

Have a look at #164. If you are able to test the change I supposed there, please test it.

Thijxx, 2017-08-06:
I'd like to test it but I don't dare messing around with the firmware at this moment..