prusa3d/Prusa-Firmware #163: 3.0.12 XYZ Calibration fails, rerun fixes it

ManuGithubSteam, 2017-08-04:
Hi

I just want to report that after the upgrade to 3.012 firmware i did also install some mods to the printer (nocuta 40mm fan) and did rerun the XYZ Calibration.

I knew it should work ok as i did not change anything else on the MK2. However the first run came up with
Calibration failed. Could not detect 4. calibration point.

I simply rerun the calibration and it worked like a charm. So it seems some limit is just barley reached and sometimes leads to the error message.

Just to inform you. If you need anything please ask.

Thanks

fiveangle, 2017-08-04:
detection is marginal if your bed is not centered in the movable area.  this is not a factor of the fw change.  redo calibration 10 times and it will probably fail again on one of them.  you likely need to move the nuts that hold the frame to the threaded rods forward or back (depending on if too far or too close to endstop limits)

PavelSindler, 2017-08-07:
It would be interesting to flesh 3.0.12 and look to menu "Support -> XYZ cal. details" and write here what data you have there. Especially first screen with "Y distance from min" would be interesting for us.

ManuGithubSteam, 2017-08-09:
These are the values:
y dist:
left: 2,01mm
Cent: 1,79
Right 1,58

GurliGebis, 2017-08-09:
Looks like the bed is not tightend the same amount in the two sides.
Or maybe the Y assembly is not all the way down in one side before the bolts where tightend to the frame (making it raised in one side of the bed).

When you did the XYZ calibration, did you let it go all the way to the top (to the point where it starts making a rattle kind of noise) ? If not, there is a chance the two Z axis are not aligned.

ManuGithubSteam, 2017-08-09:
Yes it is very hard for me to allign the frame properly. It says slightly screwd 22°
Or hat do you mean ?
Z axis goes way to the top.

PavelSindler, 2017-08-09:
Y dist values are ok. Slightly skewed is also good result. There is no need to align printer super precisely in my opinion.

closed 2018-12-28 by inquisitiveKangaroo