prusa3d/Prusa-Firmware #165: Pre-print calibration should end at the lower-left location - leaking fillament across the diagonal

Andos, 2017-08-07:
Right now the pre-print calibration checks the points in an inverse z-shape, ending at the top-right corner.
After this it moves over the print surface. 
Especially PLA filament (ABS doesn't seem to do this) leaks out across the diagonal as it moves back to the lower left corner.
Wouldn't it make more sense for it to end at that location before it prints out the priming blob down there?

Sebastianv650, 2017-08-07:
Personaly I want to have the nozzle accesible and watchable during the priming. On my MK2, this wouldn't solve the complete problem I see using PLA: It leaves small dots of PLA oozing out of the nozzle on every probe point. So the print would start on a not clean bed. The additional dots it leaves when traveling to the prime position only adds to this.

I solved that by changing the start gcode. Instead of heating the nozzle to final temperature while the bed is heating, the nozzle is only set to 160°C. Nothing will ooze at this temperature, therefore the bed stays clean. After probing, the nozzle is set to final print temperature.

So the new start gcode looks like:
```
M115 U3.0.9 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S160 ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 Z5.0 F7200.0 ; go outside print area
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 Z0.3 F7200.0
G1 X30.0 E4.5  F1000.0 ; intro line
```

As most other filaments have a higher "melt" temperature than PLA, 160°C should be quite universal as a first heat up step.

Andos, 2017-08-07:
That is a really nice idea.
As soon as the nozzle reaches the target temperature it should be ready for printing with no wait because of the oozing.
I get the small dots on the plate as well but only after the first few calibration points.