prusa3d/Prusa-Firmware #167: Feature Request: Filament runout sensor option

3d-gussner, 2017-08-08:
This is a feature request to implement 'filament runout sensor' option to the LCD settings menu.

**Why?** 
More and more people are using filament runout sensors. For the different kind of sensors we need to compile different hex files.

- Prusa default FR_SENS settings expects 'signal to gnd' when filament is detected.
- Optical, DyzeDesign sentinal and some other mechanical sensors bring 'signal to vcc' when filament is detected.

**How?**
Adding an additional 'Filament sensor' option under settings would make it possible to de-/activate '[OFF]/[ON]' FR_SENS and a second one set it to '[normal]/[invert]' the signal. Another option could be to have one menu point '[OFF]/[normal]/[invert]'
The status could be stored in EEPROM.

What have I done?
I am bit rusty in coding and got few things done, but still have some issues with it. So if somebody thinks that this is worth looking into it, i would be grateful getting some advise or help with it.

Here you can find my 1st steps to get this running: https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-FR_SENS-MENU
BTW: I changed the defualt FR_SENS pin to 'y-max' = 24.

**Possible improvements?**
- Also getting a gcode option to read and set this option, like it is available for 'Filament Sensor' (this measures the diameter)
- For the future it would be great getting FR_SENS also for the MM including a kind of 'backup' function when one filament runs out and (pre-configured by user) auto switch the filament.

Thanks

ManuGithubSteam, 2017-08-09:
Very good idea indeed. This could be adapted to work with octoprint so it sends a M command or something to octroprint so it knows that the printer detected a runout and pauses the print :-)

3d-gussner, 2017-08-10:
News: I hope i got it right, so if somebody is interested in Beta testing before i make a pull request, please check https://github.com/3d-gussner/Prusa-Firmware/commit/6b18f434995af0ae42915462df5d2677a786e41f and latest pre-release https://github.com/3d-gussner/Prusa-Firmware/releases

It would be great if somebody else could test my 'basic and rusty' code.
Thanks

GurliGebis, 2017-08-10:
@3d-gussner It looks like some line endings has been changed.
If I add the ?w=1 query string to the URL, it filters all the line ending changes away.

Have you tried running the line conversion script, so they are in the same format as the Prusa repo? :)

3d-gussner, 2017-08-10:
@GurliGebis Which line conversion script? I use notepad++ to edit my files. I am a beginner in coding and bit rusty :-)

3d-gussner, 2017-08-10:
Got somewhere between two versions a bug...have to configure it out...stay tuned

GurliGebis, 2017-08-10:
@3d-gussner The le.sh script - let me know if you want help with it :)
I can fix the line endings and make a pull request in your repo, for you to merge before you create your pull request :)

3d-gussner, 2017-08-11:
@GurliGebis thanks, found it and i will try with GIT bash to fix it :-) got also feedback from TheZeroBeast what can be improved.
Do you think this is a helpful PR? I see in the Prusa forum and thingiverse more and more people looking for a filament runout sensor. But most refer to Octoprint and not everybody has it.

GurliGebis, 2017-08-11:
@3d-gussner I like the idea, I have been thinking about adding it to my printer, but so far I have been putting it off for other projects :)

3d-gussner, 2017-08-11:
@GurliGebis I will try to get it running, wish me luck :-) just checked Marlin code and will use their updates to this topic and try get it a bit closer. Hopefully it makes it easier to adapt future changes or even to compare this function.

tkircher, 2017-08-20:
I'm definitely adding this to my printer. I recently had a spool snag and pull the filament out of the extruder fifteen hours into a sixteen hour print.

3d-gussner, 2017-08-22:
@GurliGebis and everybody
I updated the code and tested it with my signal to GND sensor, so if somebody like to review or test the code and changes i made, check here https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-FR_SENS-MENU
This firmware is based on Prusa firmware 3.0.12-3 with activated ADV_LINEAR, so make sure you add the filament specific gcode to your slicer.

GurliGebis, 2017-08-22:
@3d-gussner first of, don't get this wrong.

It looks like that tree contains 58 commits that is not upstream.
Have a look here: https://github.com/prusa3d/Prusa-Firmware/compare/MK2...3d-gussner:MK2-FR_SENS-MENU

May I suggest creating a clean pull from upstream, and make a new branch of the head of MK2, and then apply the changes needed for filament runout support.

Your tree contains changes to PID values and all other kind of stuff, which makes it near impossible to merge it.
It is always best to have the MK2 branch follow upstream completely, and then create your features as feature branches from it.
If you need it, here is a guide to keeping our own tree in sync with upstream: https://help.github.com/articles/syncing-a-fork/

Currently, it is not possible to do a diff and see what has changed, since the diff stat looks like this:

> $ git diff --stat MK2...3d-gussner/MK2-FR_SENS-MENU
 .../Arduino_1.6.x/hardware/marlin/avr/platform.txt |    2 +-
 Firmware/Configuration.h                           |   40 +-
 Firmware/ConfigurationStore.cpp                    |  914 ++++++++--------
 Firmware/Configuration_adv.h                       |   55 +-
 Firmware/Configuration_prusa.h                     |  422 ++++++++
 Firmware/Firmware.sublime-project                  |   59 --
 Firmware/Marlin.h                                  |   11 +-
 Firmware/MarlinSerial.cpp                          |   43 +-
 Firmware/MarlinSerial.h                            |   59 +-
 Firmware/Marlin_main.cpp                           |  196 +++-
 Firmware/cardreader.cpp                            |  471 +++++++--
 Firmware/cardreader.h                              |   61 ++
 Firmware/langtool.pl                               |    4 +-
 Firmware/language_all.cpp                          | 1112 +++++++++++++++-----
 Firmware/language_all.h                            |   47 +-
 Firmware/language_common.h                         |    3 +
 Firmware/language_de.h                             |  618 +++++------
 Firmware/language_en.h                             |  131 ++-
 Firmware/language_nl.h                             |  331 ++++++
 Firmware/mesh_bed_calibration.cpp                  |    5 +-
 Firmware/pins.h                                    |   16 +-
 Firmware/planner.cpp                               |   66 +-
 Firmware/stepper.cpp                               |   14 +-
 Firmware/temperature.cpp                           |    6 +
 Firmware/ultralcd.cpp                              |  260 ++++-
 Firmware/ultralcd.h                                |   12 +-
 Firmware/ultralcd_implementation_hitachi_HD44780.h |   26 +-
 Firmware/variants/1_75mm_MK2-RAMBo10a-E3Dv6full.h  |   10 +-
 Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h  |  823 ++++++++-------
 29 files changed, 4023 insertions(+), 1794 deletions(-)

Some of these changes are not related to the filament sensor feature :)

3d-gussner, 2017-08-23:
@GurliGebis Thanks for the answer. I made few changes other the time and had just one or two branches, but as you say maybe it is better to make different branches.
Changes i made:
- baud rate from 115200 to 250000
- modified language.en so using the language.pl perl script doesn't send that much warnings
- translation to German, which was pulled by Prusa
- translation to Dutch, maybe i have to try again to get it pulled
- added PID tuning for the heated bed to LCD menu...useful if you use cork underneath
- to make it bit easier for me i added my PID settings for the hotend and **bed**
- changed Preheat order 1st PLA, 2nd ABS...i print most times with PLA and like to preheat the printer
- removed unused E_MUX2_PIN 84
- DISABLE_MAX_ENDSTOPS
- activated LIN_ADVANCE, what a great feature thanks to @Sebastianv650 and everybody
- and last started with filament runout sensor
  - 1st version was just activating it and provide hex files for users not comfortable compiling their own firmware 
  - changing the PIN from 21 to 24 (y-max)
  - modified change position from x211, y0 and z-add 2 to x0, y-2.2 and z-add 20
  - added filament runout sensor to LCD screen. Thanks to @TheZeroBeast helping with optical sensor and coding
  - modified filament runout sensor terms and code closer to Marlin

It will take bit time to make all branches and changes. Thanks again for pointing out!

GurliGebis, 2017-08-23:
It makes it a lot easier to review and merge, if it is split up.
You can always have a combined branch where they are merged, that you use to test it :)

3d-gussner, 2017-08-23:
So took a while to split everything, hope didn't mess up
@GurliGebis hope i did it better now
- Filament Runout Sensor check https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-FRS 
- German translation update check https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-DE [pull request opened]
- Cleanup check https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-Cleanup [pull request opened]
- 250000 baud rate check https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-250000_baud [pull request opened]
- PID tunig for heated bed check https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-PID_BED

next will be updating the Dutch translation

GurliGebis, 2017-08-23:
@3d-gussner I haven't got a sensor to test it, but it looks great from what I can see.
Would need someone like @PavelSindler to take a look at it though :)

GurliGebis, 2017-08-23:
You might want to create a pull request for that branch :)

3d-gussner, 2017-08-24:
@GurliGebis Hi, we will see if @PavelSindler and Prusa like that idea. Thanks for 'quick' review/opinion of the code we wrote.
I will wait with the pull request until i get some feedback from @TheZeroBeast, who helped testing and coding with an optical sensor, or somebody else willing to test it.
More and more new printers have a filament sensor and in each review it is THE THING, so hopefully Prusa will add this to their firmware.

TheZeroBeast, 2017-08-24:
Hi Gussner,

I’ll see if I can get the firmware loaded in the next day and report back if I find any bugs.

I saw your posts on the PINDA sensor drift as well and have been thinking about a way to make it more fool proof.

Anyway, have a good day mate.

3d-gussner, 2017-08-24:
@TheZeroBeast Hi, as you can see i split to many branches to make it easier to review. I will have a branch containing all changes i made.
The PINDA thing is quite difficult and the preheat attempt Prusa made, is working quite okay if you print 24/7. But the subject is quite tricky because the thermal issues with the bed, PINDA and hotend.
The PINDA is great to detect for mesh bed leveling but for the height you gonna need something that physical detects the distance between the bed and nozzle. I like the Dagoma Neva approach, still don't know how exactly they detect the height but guess something like a piezo sensor is used. An other idea would be using TMC stepper drivers which can 'detect' forces on the stepper motors. This is used in Marlin to get home position without mechanical/optical switches. But that is another story, the Prusa i3 MK2(s) doesn't have a controller with TMC 2130/5130 stepper drivers.

You also have a nice day.

3d-gussner, 2017-08-30:
Hi @PavelSindler,

@TheZeroBeast and i tested https://github.com/3d-gussner/Prusa-Firmware/tree/MK2-FRS and got good results with two types of filament runout sensors.
Should we try to make a pull request or is it something Prusa isn't interested in?

3d-gussner, 2017-10-07:
@PavelSindler I know you are busy, but seeing the filament sensor on the MK3 I assume that Prusa will add this feature also to MK2.5 upgarded printers. So does it make sense to update my/our (@TheZeroBeast) branch to 3.0.1-RC1 and try a pull request, or are you already busy with the implementation for the MK3/2.5?

GurliGebis, 2017-10-07:
I think they already have it in, since it is working on the ones Prusa has demoed. But it seems like it is running code currently in a private repo (most likely going live once the first unit ships, like the STL files)

closed 2018-12-28 by inquisitiveKangaroo