prusa3d/Prusa-Firmware #168: Stop print sometimes requires reset

tkircher, 2017-08-10:
Since upgrading to 3.0.12, sometimes when I stop a print manually, the printer continues the current motion until it hits the end of the rail, and tries to keep going. It apparently won't stop on its own, and requires a reset.

GurliGebis, 2017-08-11:
I have seen something related to this, where I tried telling the extruder to extrude 100mm filament, but it kept going untill I reset it.
I haven't been able to reproduce it though.

PavelSindler, 2017-08-16:
Thank you for reporting this. Does the problem with stop print occured during printing or was it stopped at the beginning when preheating, homing or mesh bed leveling takes place?

GurliGebis, 2017-08-16:
I can only speak for my issue, but what I was doing was the extruder calibration, where I asked it to extrude 100mm of filament - I cut the power when it reached around 150mm :)

tkircher, 2017-08-20:
The problem occurred during printing. I managed to reproduce it a few times, always when it was in the middle of printing a straight line. There was one time I also saw a message on the display, something like 'waiting to finish'.

GurliGebis, 2017-08-20:
Yesterday I saw something like this too, although not entirely.
I had just changed to another roll of filament and started a print, where the purge strip was printed at around double length - then the print started normally.
Had to cancel it and start over (forgot infill, dohhh), this time it did it normally.

Not sure if it is related, but thought I might as well report it :)

closed 2018-12-28 by inquisitiveKangaroo