prusa3d/Prusa-Firmware #169: Files on SD card: The sorting is wrong

m-Th, 2017-08-11:
On Windows we have the following sorting (note: πμ. = AM μμ. = PM in Greek) which is also the correct one:

[image]

But on the Prusa's LCD the files are displayed like this (starting from the 1st one (!) which, as you see above, it should be the last one):

[image]

Can you fix the sorting?

PavelSindler, 2017-08-14:
There were currently no real sorting in SD card menu (files were shown in same order as in FAT directory tables). In most cases files which were created on SD card recently were shown on the top of SD card menu. The oldest files were shown in the end of SD card menu. But if you removed some files and then add new file, this didn't work any more.
There are many different options for sorting. Sorting which you have on top figure is possible but I think that it is better to sort by time when file was created on SD card (this means also copied to SD card). If you have some gcode which is really old, but you have just copied it to SD card, you probably want to print it and thus it should be on the top of SD card menu. 
I am currently working on SD card menu sorting and it will be available in next release. Defaultly files will be sorted by time, when they were created on SD card. There will be also option to sort alphabeticaly and option to turn whole sorting off.

m-Th, 2017-08-15:
Thanks! Saw the pull request. Btw, don't forget to honor M34 code: http://reprap.org/wiki/G-code#M34:_Set_SD_file_sorting_options

cjparsons, 2017-09-12:
Excellent news - thank you very much, I look forward to the next release

I had raised this before (probably in the wrong place) but I think your changes will be very helpful

joseamirandavelez, 2018-02-14:
Is the sorting done using the creation date, or modification date of the file? I noticed that files that are overwritten directly in the SD card do not get to the top of the list if there is a file with a newer creation date.

cjparsons, 2018-02-15:
I never use the sort by date option - the listing is never correct...

The top file in the list is never the most recently changed or copied, I have quite a few files on the SD card which may cause the sort to fail - but the alphabetical sort _does_ appear to work?

Not sure if it is relevant but my SD card is a wireless card - a 'FlashAir' - might try copying the files to a standard card and see if that makes any difference sometime