prusa3d/Prusa-Firmware #170: Pausing print then changing filament causes the layers to shift

mrjohnc, 2017-08-12:
Pausing the print and then changing the filament does something where the printhead starts again forward of where it is supposed to start 

Latest firmware, latest Prusa Slic3r

[image]

3d-gussner, 2017-08-23:
If you want to change the filament during the print and you are not using gcode to do that, then there is a 'Change filament' option under 'Tune' during the print. Maybe that helps. Advantage of this that the nozzle heat stays at set temperature, while pause will set nozzle temp to 0!

You should try PrusaControl, even it is still a beta, but has a nice feature to add filament changes, even if you import gcode.

BTW
I tried to pause, unload filament, tune (nozzle temp to 210 because it was dropping), load filament and resume print and it worked without layer shifting. I use fw 3.0.12-3

mrjohnc, 2017-08-23:
Thanks, I used the change filament option to do this. I'm not sure why we are getting different results....

3d-gussner, 2017-08-24:
That is really a strange! Which fw version do you have? Does it happens all the time or sometimes?

closed 2018-12-11 by inquisitiveKangaroo