prusa3d/Prusa-Firmware #171: Pausing the print then changing the filament causes the printer to think the target temperature is 0

mrjohnc, 2017-08-12:
Pausing the print then changing the filament causes the printer to think the target temperature is 0 when resuming a print. The printer will attempt to print with a temperature much lower than needed to extrude filament.

Latest firmware, latest Prusa Slic3r

PavelSindler, 2017-08-14:
Can you please provide more information? What procedure did you use for changing filament?

mrjohnc, 2017-08-14:
Upload filament + load filament, I think I only used the unload filament and load filament commands once each, but it could possibly be twice

PavelSindler, 2017-08-14:
I've just tried few scenarios like that and I have no problem with target temperature resuming so far.
Isn't it just some misunderstanding? Are you aware that during pause print target temperature is set to 0 and when you choose "Resume print" in main menu, stored target temperature is set back to proper value?

mrjohnc, 2017-08-14:
Are you saying I have to manually set the temperature of the hot end back after I've paused a print before I resume it? That can't be right?

m-Th, 2017-08-15:
> Are you saying I have to manually set the temperature of the hot end back after I've paused a print before I resume it? That can't be right?

No.

He sez like this:

1. You're printing at 215 degC (for example)
2. You press "Pause Print"
3. The firmware sets the target temp to 0 degC because keeping the plastic hot for too long will destroy it.
4. You unolad/load filaments, take a break, a coffee etc. The temp falls down.
5. After you did everything at step 4, you MUST press RESUME PRINT. This is to be found in the same menu with Pause Print. Otherwise the printing process will not continue. When you press Resume Print, the printer will retrieve the previous temperature (which was saved at step 2 in the printer's memory) and the printer will wait till the hotend will reach again this value. Only then, the printer will resume printing.

PavelSindler, 2017-08-15:
Yes, it works as @m-Th described. Setting nozzle target temperature to zero during pause also means that printer is quiet during pause (when nozzle temp. drops under 50 C, both fans are turned off).

mrjohnc, 2017-08-16:
Ok, please retire this request then. Thanks very much for the explanation

closed 2017-08-16 by PavelSindler