prusa3d/Prusa-Firmware #172: Filament change menu should include the unload option

bagobones, 2017-08-13:
I find that the filament sometimes does not unload correctly during the filament change... There is no option to try again if it has failed when using the built in change command. Currently it only includes the yes complete and load again option.

It would be preferable to run unload again (it happens once automatically at the beginning of the change) than to manually pull open the tension or even unscrew the tension screws and reassemble just to do a simple change.

Naugrimohtar, 2017-08-14:
I agree. Is there a way to upvote?

mrjohnc, 2017-08-16:
+1 also

ChiefPoints, 2017-08-20:
I only recently started to use the color change ability and this definitely is a must have feature. All to often the filament fails to fully eject on the first run and you are left with the choice of loosening the tension or forcibly pulling the filament up.

tkircher, 2017-08-31:
Yes, we definitely need this option.

Naugrimohtar, 2023-08-14:
+1 for this

bagobones, 2023-08-14:
Given the age of this request I am guessing I will not see this for my old MK2s if it ever gets implemented.

This is quite an old request at this point.. Surprised more people where not very frustrated with it.

closed 2023-10-24 by github-actions[bot]