prusa3d/Prusa-Firmware #178: "Stop Printing" forces us to unload at least one filament. Please change this

m-Th, 2017-08-21:
When we choose "Stop Printing" a menu appears saying "Unload all"/"Unload current"/"Unload printing filament".

Please add an option saying just "Exit" (or similar). We want to keep all the filaments loaded, just to stop the printing.

domesticatedviking, 2017-08-23:
I agree 100%.  Unloading and cutting filament on a single extrusion print that has to be aborted is wasteful, tedious, and unnecessary.

3d-gussner, 2017-08-23:
Question: You are talking about a Prusa i3 MK2(s) Multi Material? Because this doesn't happen on a MK2s SM.

m-Th, 2017-08-23:
@3d-gussner : in title I said "...**at least one** filament". In the issue's body I quoted the menu with "Unload all/Unload current/Unload printing filament". 

From all these one can infer that indeed I am speaking about MK2S MM. :)

ceryen, 2017-09-09:
I agree with this request.  I would like the option of simply pulling the currently extruded filament back into the multiplexer rather than fully unloading the filament.  Whenever I'm stopping a print, there's a very good chance I'm going to restart the same print, so I'd like to keep the same filament loaded.

jokohoko, 2017-09-25:
Also agree, i seem to be using the reset button way to often when trying to make the multi material to work. (and resetting is mostly to skip the unload procedure.

macfly1202, 2017-09-25:
Completely disappointed with this behavior. Why must we unload one or more filament COMPLETELY? why not unload 5 cm only? It would be great if this option - must often use option I say - could be added to the menu.

Burner911, 2017-09-30:
I agree, I would also appreciate an option of being able to stop a print without having to completely remove at least one material.

henrikkoch, 2017-09-30:
totally agree I have mentioned this to Prusa - They didn’t react on this at all

Med venlig hilsen
Henrik J. Koch

edunad, 2017-10-22:
+1 this

Quite annoying, i had to stop the print because the model slided :P, it wasn't because of filament being stuck

closed 2019-04-11 by michalxfanta