prusa3d/Prusa-Firmware #180: Calibrate Z drives hot end into heat bed when not homed

clearbucketLabs, 2017-08-22:
When calibrating Z from the calibrate menu it should home X/Y first.  i don't remember what i was doing in particular but i had the X in the center and back of the Y with the Z maybe 20mm up.   when i did calibrate Z, it just moved down and didn't stop and driven itself into the bed to the point of skipping Z motors.    this nearly damaged the print bed given how far it bent.

tkircher, 2017-08-31:
When it asked you to move the Z axis all the way up to the top, did you do it?

clearbucketLabs, 2017-08-31:
Yep,  it's a bit of an oddity.    i guess it wouldn't matter if it's homed if it uses the Z height to stop.

tkircher, 2017-08-31:
There should definitely be some kind of redundancy for the Z axis. I had a PINDA probe fail due to cable fatigue, and the print head marred the PEI before I could hit the reset button. After cutting out the fault, soldering it back together, and replacing the hard plastic cable sleeve with mesh, I never had that problem again. But a simple switch as a failsafe would solve many problems. The distance between the print bed and the nozzle, relative to any point on the frame, can only ever vary by some fraction of a millimeter. Even with an extreme negative Z offset, in the worst possible case it wouldn't visibly deflect the print bed before it trips the limit switch.

lenalebt, 2017-09-16:
The same is true for calibrate XYZ - although I do not see any tricks to stop it from driving into the bed for calibrate XYZ. Pinda probe is working though - for example, auto home works, and "show end stoppers" in "calibration" also shows a working probe. The LED goes off as well, it nevertheless keeps driving into the bed...

paulwoods, 2017-09-18:
I have the same issue. Here is a video: https://www.youtube.com/watch?v=2e49Ga9Kqys. My MK2 would do this about 4 out of 5 times. The 5th time, it would calibrate. I've done the MM update today, and now it won't find the 2nd probe point. Its seems to be about 10mm below the probe point, and eventually, starts to scratch into the PEI.

tkircher, 2017-09-18:
Paul, the Z nuts on your printer are upside down. So the Z axis can't go all the way up, so the Z offset is wrong.

tkircher, 2017-09-18:
This is exactly why there should be a redundant Z axis endstop, though. The endstop is useless for calibrating the print bed, but provides a high degree of crash safety. The PINDA probe offers high precision and an absolute offset from the print bed, but relying on it exclusively offers zero safety.

paulwoods, 2017-09-18:
I'm an idiot. I just noticed this: before, when I calibrated, i would home first. Now, I just move the head to x-center then calibrate. seems to work now.

GurliGebis, 2017-09-18:
@paulwoods Your problem is related to these two being turned upside down:

https://d17kynu4zpq5hy.cloudfront.net/igi/prusa3d/kICaEEPNA1S2QDGl

They are inverted on your printer, blocking it from going all the way up, and therefor it thinks it is higher than it is, when you press yes to it being all the way up.

paulwoods, 2017-09-19:
Thanks, all. Correcting those brackets seems to have fixed my issue.

lenalebt, 2017-09-20:
My problem looked similar, but was caused by a wrongly mounted hotend. Mine was mounted about 1cm too low. Nevertheless, I would expect the firmware to stop driving towards the bed as soon as the pinda probe says "there's the bed", which did not happen. The probe has been mounted below the nozzle, so it should have stopped before driving into the bed, right?

paulwoods, 2017-09-20:
I'm guessing that the nozzle goes down a fixed distance. I thought before that it was using the pinda probe to find the bed as it was going down, but now I dont think so. Now, i understand @tkircher comment about a mechanical z-min end-stop. It would help keep this from happening...

tkircher, 2017-09-21:
Yep, exactly. There's already a connector for a Z endstop on the control board. The cost of an additional endstop switch is pretty low, and it would be easy to design in. Certainly worth it for the peace of mind of reducing the possibility of a head crash to almost nothing.

Crunch69, 2018-04-05:
this happens intermittently if you do not home first before starting a print

closed 2018-12-28 by inquisitiveKangaroo