prusa3d/Prusa-Firmware #185: add quick access menu

TonyLHansen, 2017-08-23:
I've often thought that there needs to be a "quick access" top-level menu that offers a few simple commonly-used items that are otherwise buried rather deep. For example, "move Z up 5cm". (Yes I know that there is no buried menu item that says "move Z up 5cm", but that is something I find myself wanting to do quite often.) The items that are available on this "quick access" menu should be individually choosable.

3d-gussner, 2017-08-24:
I guess coding this personalized menu would take a lot of effort and still not everybody would be able to get what they want.
You have to remember that this controller uses a atmega 2560 8-bit chip with LIMITED CPU power and ressources and its primary function is translating gcode to as far as possible accurate physical movements and print results.
If lot of people could agree to ONE version of quick access, that could be done.

ceryen, 2017-08-25:
This isn't exactly what you asked for, but here's a useful tip: if you long-press the button, it'll jump into the Z-axis position screen, which will allow you to quickly move the Z-axis

tkircher, 2017-08-31:
It would be easy to set up a Raspberry Pi with a touchscreen that interfaces over USB, if you want more sophisticated control right at the printer.

clearbucketLabs, 2017-09-08:
The menu could just be based on SD Card gcode in some folder.  like "quickaccess" and the name of the menu item is the name of the gcode minus extension.  it would be similar to just going to print from SD card but i guess filtered for just those operations so you don't have to scroll through your print files.  only disadvantage would be it's on the SD card, if you have more than one then it would have to be cloned to others.

inquisitiveKangaroo, 2018-12-28:
Move Z-axis shortcut has been implemented.

closed 2018-12-28 by inquisitiveKangaroo