prusa3d/Prusa-Firmware #189: [Request MMU] Pull current filament into cooling tubes after every print.

HonskiDesign, 2017-08-31:
Hello,

This is an issue that I've noticed over the last couple months of using the MMU.

After a print on the MMU, the filament last used stays 'Loaded' at the extruder. If I decide to start another print immediately after with a different color from a different tube, it will try to drive the new filament into the extruder where the previous filament was left causing a jam.

The simple fix is to add a retract to the ending script in Slic3r(or whatever slicer the user chooses) so that the filament is pulled back into the staging area inside the cooling tubes(steel tubes in the multiplexer).

Another reason why this is needed is because before every print it primes the extruder(lays down filament) in a large quantity across the front of the build plate. If the filament is already 'loaded' and not 'staged' back in the steel tubes, it causes the extruder motor to click and bind as it can drive that much filament through the extruder.

Let me know if you need any clarification.

PavelSindler, 2017-09-01:
Hello, there should be pulling filament into cooling tube in print end sequence. Which slicer are you using? Can you send gcode?

HonskiDesign, 2017-09-01:
Using the latest Slic3r MultiMaterial. Here is a file before I added it to the ending script.

[MaleInsertBody_Nylon.txt](https://github.com/prusa3d/Prusa-Firmware/files/1269907/MaleInsertBody_Nylon.txt)

mavu, 2017-10-06:
on my MMU MK2 everything is correctly retracted after a Multimaterial print. (latest released firmware and slic3r)

If you use Slic3r in MMU Single mode, the firmware asks after the print to unload the filament. As fas as I can tell you can only get to the situation you described if you ignore the message on the display and switch the printer off without unloading.

FelisCorax, 2017-10-16:
I'm having the same behavior on my Prusa i3 mk2s with MMU. I'm using slic3r prusa edition with the included MM print profile, and using octoprint on a raspberry pi to control the printer. Sofware is up-to-date as of about a week ago when I installed the upgrade.

With an actually multi-material print the last filament type is retracted correctly no problem, but if I'm only using a single material type then the last material is left in place when the print finishes (there's no prompt either on the prusa display or on octoprint). The solution I've come up with is to use the control tab on octoprint to extrude a little bit of filament to clear any lumps and then retract it back to about where it ought to be. In other words, I'm manually giving the printer instructions that probably ought to just be in the gcode.

That said, I've just noticed this is the github for the firmware, not for slic3r, so this may not be the right place to request this particular fix.

catrielmuller, 2017-10-16:
I think the problem is only if you use the MM (Not Single Mode) for print one single color. But the Single Mode Profile have the Gcode for retract the current filament to the cold tube.
For resolve this issue you can put this GCODE into the OctoPrint in "After print job is cancelled" and "After print job completes"
```
G1 E-4 F2100.00000
G91
G1 Z1 F7200.000
G90
G1 X245 Y1
G1 X240 E4
G1 F4000
G1 X190 E2.7  
G1 F4600
G1 X110 E2.8
G1 F5200
G1 X40 E3  
G1 E-15.0000 F5000
G1 E-50.0000 F5400
G1 E-15.0000 F3000
G1 E-12.0000 F2000
G1 F1600
G1 X0 Y1 E3.0000
G1 X50 Y1 E-5.0000
G1 F2000
G1 X0 Y1 E5.0000
G1 X50 Y1 E-5.0000
G1 F2400
G1 X0 Y1 E5.0000
G1 X50 Y1 E-5.0000
G1 F2400
G1 X0 Y1 E5.0000
G1 X50 Y1 E-3.0000
G4 S0
M107 ; fan off
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
G28 X0  ; home X axis
M84     ; disable motors
```
if you put this into "After print job completes" probably you need remove these lines:
```
M107 ; fan off
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
G28 X0  ; home X axis
M84     ; disable motors
```
from you end script in the profile of Slic3r.

closed 2019-01-03 by inquisitiveKangaroo