prusa3d/Prusa-Firmware #192: The progress percentage is way off.

m-Th, 2017-09-02:
With 3.0.12 the progress percentage (SD: % indicator on printer's LCD) is way off at least on some prints. How it is calculated?

On the old firmware editions it was calculated (ISTM) as a percentage of Z height. 

A test case is to be seen with https://www.thingiverse.com/thing:659066 . 

Settings used:
Infill: 100% Rectilinear, Layer height: 0.2
Extrusion: 1.5 (FormFutura HDGlass filament)

Printer's LCD says 46%

Z height is somewhere at 58%. The thing is printed rotated like in the photo below (LCD has the colors enhanced), so the upper part is significantly smaller. A percentage indicator based on volume should say at least 80%. However it says 46%. 

Why? 

[image]

GurliGebis, 2017-09-02:
AFAIK it is calculated on how far along in the file it is.
So if the gcode is 100.000 lines, 1.000 lines in would be 1% on the display.
The problem is, that even though being 50% through the file does not mean that the time it takes is half way done, just that half of the commands needed to do the print has been processed.

If for example the model is more complex at the bottom, that part would contain more gcodes than if the top is a more simple structure.

Hope that makes sense :)

m-Th, 2017-09-03:
Well, YOUR explanation makes sense, thanks. :)

However I really don't know how much sense the implementation has in this way.

A better approach IMHO would be to measure the length/volume of filament already spent. The percentage will come from the total length/volume needed (already stored in the Slic3r) divided by the amount used.

lekoder, 2017-09-03:
> A better approach IMHO would be to measure the length/volume of filament already spent. The percentage will come from the total length/volume needed (already stored in the Slic3r) divided by the amount used.

That's not trivial to implement on firmware side. Perhaps it could be done by writing special g-codes in slic3r, which would instruct printer to display specific information on LCD?

m-Th, 2017-09-03:
Yes. It is already there. 

 

In order to see it, take the following steps:

 

1.       Slice a model with Slic3r PE (Prusa Edition)

2.       Open the gcode file with a text editor and go (almost) to end

3.       The file ends with an epilogue like this: (with „//” are my comments)

a.       M107

b.      G4 ; wait

c.       M104 S0 ; turn off temperature //---- normal housekeeping

d.      M140 S0 ; turn off heatbed  //--- close all

e.      M107 ; turn off fan  //--- fan not needed anymore

f.        G1 X0 Y200; home X axis // --- ok, go home

g.       M84 ; disable motors // --- we are all done

h.      ; filament used = 3182.9mm (7.7cm3)  //< ---- THIS! Here is the info we are looking for!

i.         ; total filament cost = 0.0

j.         

k.       ; avoid_crossing_perimeters = 0  // --- from now on, Slic3r stores the settings used to slice the model

l.         ; bed_shape = 0x0,250x0,250x210,0x210

m.    ; bed_temperature = 55

n.      …

 

As you see above, the critical line is 3.h. You can do a simple text search for a line beginning with „filament used = ” and take the total length value from there.

 

The code for the actual (current) value used in the current print is surfaced in „Statistics” menu. You need to divide these values in order to get the percentage.

ceryen, 2017-09-08:
Note that filament used isn't an accurate measurement of time, so the percentage reported will still seem arbitrary to the user.

For example, the first half of a print may consist of long lines where the nozzle can accelerate to maximum velocity.  The second half of a print may consist of short lines where the nozzle has to frequently accelerate and decelerate, and where the nozzle cannot accelerate to maximum velocity.  The second half would take way longer, even if it uses the same amount of filament as the first half.

laird, 2017-09-08:
The Sailfish firmware (a fork of Marlin, used on Replicator-style printers mainly( has packaged the acceleration logic of the firmware into a tool that can be run on gcode in order to generate the precise print time. It was created as a debugging tool, I believe. I wonder if it might be reasonable to do the same thing for the firmware in the Prusa, which would allow slicers to report precise print times. (For a given set of firmware parameters for acceleration, jerk, etc., of course.).

closed 2018-12-28 by inquisitiveKangaroo