prusa3d/Prusa-Firmware #194: Minimize oozing filament preprint with controlled heating of nozzle

JohnRThomas, 2017-09-07:
I've noticed that the nozzle gets up to temp a full minute or two before the bed which causes an excessive amount of filament to ooze out of the nozzle. If this isn't caught and removed before the first layer, this can cause the first layer to be pulled off be dangling filament. 

A simple solution would be to start heating both like normal, but then limit the rate of increase that the nozzle is going up by so that it reaches full temp at about the same time as the bed.

GurliGebis, 2017-09-08:
It has been discussed before, most recently here: https://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/dealing-with-ooze-during-mesh-bed-leveling--t5582.html

ManuGithubSteam, 2017-09-22:
This is a good idea, Prusa should implement :-)

closed 2023-08-24 by github-actions[bot]

JohnRThomas, 2023-10-09:
I was able to implement this with custom startup gcode in the the slicer. PrusaSlicer allows you to override the temp code, I just flipped the two lines so that the bed is headed before the nozzle.