prusa3d/Prusa-Firmware #196: check if gcode file is complete

Toifur, 2017-09-10:
It would be great if the gcode will be checked for completeness and integrity before the actual print starts.

GurliGebis, 2017-09-10:
What would define it as complete?

Toifur, 2017-09-11:
Maybe an extra line of code in Slic3er PE generated files or a  "M84 ; disable motors"  in general.

It could help to avoid half done prints caused by an overhasty removed SD Card.

GurliGebis, 2017-09-11:
But that would require all other slicers to do the same thing, and force people to end their gcode like that (the stop gcode can be changed by the user).

probonopd, 2017-09-11:
This sounds to me like a generic feature request for upstream Marlin in general and not Prusa specific. What could be done is to check whether the final position of the extruder is further away from the last position where extruding was done, and show a warning if the two are too close or identical. I am not sure how relevant the use case is and whether it would justify the effort.

Do you often have incomplete files @Toifur and how do these happen?

PavelSindler, 2017-09-13:
Hello, yes I am currently working on it and it should be in next release. I am checking M84 gcode in end section of file, which in my opinion should work with most slicers and it is probably better approach then adding some prusa specific gcode to end section. If file is not complete, there will be some warning message, but could continue anyway. Incomplete files are quite frequent problem. Usually user removes SD card during slicing.

bubnikv, 2018-10-11:
This has been implemented long ago.

closed 2018-12-28 by inquisitiveKangaroo