prusa3d/Prusa-Firmware #197: M600 does unwanted extrude after 'is color clear?' confirmation

jwveldhuis, 2017-09-12:
See http://shop.prusa3d.com/forum/print-tips-slic3r-settings-kisslicer-model-repair--f12/colorprint-color-change-spews-additional-after-col-t5790.html#p41430

After changing the filament, cleaning the nozzle and confirming the 'color is clear', it starts extruding a bit and then continues to print. So you need to be very quick in catching that piece of hanging filament before it destroys the print.
Better would be to do a small retract.

GurliGebis, 2017-09-12:
Are you sure it is extruding it, or is it just oozing?

jwveldhuis, 2017-09-13:
It is definitely extruding, can hear the stepper and it is extruding about 2-3cm of filament just before the hotend moves back to the original position.

mloidl, 2017-09-13:
@jwveldhuis 
This is the desired behavior to stabilize the pressure.

See code below which is executed after user confirmed the filament change:
```
       //Not let's go back to print
 
       //Feed a little of filament to stabilize pressure
       target[E_AXIS]+= FILAMENTCHANGE_RECFEED;
       plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], FILAMENTCHANGE_EXFEED, active_extruder);
 
       //Retract
       target[E_AXIS]+= FILAMENTCHANGE_FIRSTRETRACT;
       plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], FILAMENTCHANGE_RFEED, active_extruder);
```

jwveldhuis, 2017-09-15:
Why can't that be executed before the user confirms the color is clear? Now it throws out filament and you really need to rush to catch the additional filament while the hotend is moving already..

mloidl, 2017-09-16:
There might be a delay between loading the new filament and the confirmation from the user. Depending on the time of the delay this can cause some oozing which lead to a maybe non stabilized filament stream inside the noozle.

If you know that there is some filament extruded after confirmation you can be prepared and should be fast enough.

My solution is following
* Switch filament
* When question is shown take a plier or position my finger under the extruder
* Confirm the filament change with the left hand
* Wait until the filament is extruded and take it with the right hand.

Area5142, 2018-05-30:
I print a lot of PETG and this extra extrusion gives 8-10mm ooze before the nozzle reach its resume point on the model, typically sticking to the perimeter on the way...
Very hard to catch if color change is at low layer height.

But is this still necessary after Linear Advance (LA) is part of the firmware?
LA should maintain the right pressure defined by M900 Kxx - even build up the right pressure as the Kxx parameter specifies what is needed.

If LA is disabled, the old "extra extrude" could still be the default action.

(Have checked latest firmware 3.2.1, this extra extrusion code is still in Marlin_main.cpp)

3d-gussner, 2018-05-30:
@Area5142 I think that the extra purge isn't that bad as if somebody walks away during this process the 'nasty' molten plastic gets out directly before the print.
It is bit annoying but i got used to it and just use tweezers when pressing to confirm.
Below layer 10 it moves +10 layers above up to give some clearance to clean up.

Kadah, 2018-12-29:
Just used M600 in a print today and ran in to this.

The behavior is very much non-intuitive unless you already know its going to do this. Its already purged, confirm changed, the print head starts to move to resume and suddenly it spews out 4-6cm of filament while moving in to the print.

I didn't even notice the first time till I saw a 6cm loop of loose filament half printed over in to the current layer. The fan had pushed the this extra extrusion backwards, out of view under the print head, after finishing M600. It ruined the print, but luckily this was only a tiny test object.

The issue is that the cause of it is non-obvious without reading the source (or googling to find #904 then this one). I had started by looking at the gcode to see what was happening, but nothing obvious there to cause such a large amount of extrusion post-M600.

Pathogenius, 2018-12-29:
I agree that the M600 routine could use some polishing. It would be useful at the very least if would do a routine where it printed a test line repeatedly and ask each time if 'ready to print?'. I've always had to use a chopstick to try and grab the scrap out of the way when it starts, which can be tricky.

St0RM53, 2019-09-20:
I made a post in the forums regarding this issue because i am doing multimaterial multicolor prints with a stock mk3s (no mmu, single extruder)
Post with details and video here: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/m600-is-broken/
Is this ever gonna be "fixed"?

St0RM53, 2019-12-17:
Still not fixed?!?

GilesBathgate, 2020-05-08:
What happens if you just set the `FILAMENTCHANGE_RECFEED` to 1?

```c
#define FILAMENTCHANGE_RECFEED 1
```

bugraceylan, 2021-08-24:
This is really annoying, we're arranging competitions in office to get excess filament before extruder reaches the part

GilesBathgate, 2021-08-24:
Given the comments, I don't think this is likely to be "fixed", because it seems to be intentional. 
Just remove this code if it annoys you.

```c++
      //Feed a little of filament to stabilize pressure
      target[E_AXIS]+= FILAMENTCHANGE_RECFEED;
      plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], FILAMENTCHANGE_EXFEED, active_extruder);
```
It is intentionally annoying ;)

Area5142, 2021-08-24:
With all the comments on this _"extra feed"_ on color change M600 gcode, I think most users find it very annoying (including myself). I suggest extending the M600 gcode with an extra parameter, specifying the extra extruder length to _"build up pressure"_:

`M600 Lx ; Color change with extra extrude of x mm`

`M600 L ; Color change with extra extrude defined by firmware (as the gcode is today)`

`M600 ; Color change without extra extrude`

I know that this will change the default behavior of M600 to not extrude extra filament on change, but it also makes it possible to have the extra extrude if anyone wants it.

GilesBathgate, 2021-08-24:
@Area5142 Alternatively `M600 L0; Color change without extra extrude` and leave current behaviour as is.

gudnimg, 2021-08-24:
Hi everyone, just wanted to mention I created a PR with changes related to M600: https://github.com/prusa3d/Prusa-Firmware/pull/3248 perhaps it is related to the discussion?

GilesBathgate, 2021-08-24:
@gudnimg It probably is a bug that it does the filament change final feed twice, but only insomuch that twice the amount of filament is extruded than intended.

Area5142, 2021-08-24:
@GilesBathgate I didn't check for already used parameters - it could instead be P for post-extrude.
To not break any already made gcode, the default behavior could be as today (firmware defined extra extrude) and the P could specify extra length in mm.

```
M600 ; Firmware defined extra extrude as today
M600 P0 ; No extra extrude
M600 P1 ; 1 mm extra extrude
```

puterboy, 2023-04-27:
This is really annoying and despite several suggestions mentioned over the years, it still has not been fixed.
I like the suggestion by @GilesBathgate

> M600 ; Firmware defined extra extrude as today
> M600 P0 ; No extra extrude
> M600 P1 ; 1 mm extra extrude

But if you don't want to change the M600 code, then how about at least inserting a 2 second pause after the extra extrusion to give the user time to wipe the stray extruded filament. Maybe even a beep and/or a screen message to remind/prompt the user.

Relying on the user to have precise timing and coordination to wipe the extruded filament as the extruder is descending and moving towards the object seems to be a very user-unfriendly and unreliable method.

If you make a mistake and either fail to fully wipe the extruded filament or you drop it on the print, you risk ruining a potentially multi-hour print with a stray goober or line that adheres to the print. These gymnastics are the worst part of inserting a filament change or other pause to my prints.

This seems to be the opposite of good UI/UX design -- especially since several workable solutions have been suggested over the 6 years since this bug was first posted!

Prusa-Support, 2023-06-28:
Thank you again for your input.

I'd like to clarify this is intended behavior to prevent gaps due to excess oozing that may cause an empty spot on the layer.
Priming filament is especially useful in case the user doesn't resume the print right away, and makes it up for that scenario.

The current behavior may cause small artifacts like extra filament outside the object but no gaps, in fact, there is a chance that the user will let the nozzle ooze for some time before resuming and there would be no work around that.

Our developers will make their considerations about this.

For now, please let the nozzle ooze for several seconds before picking the purged plastic (and resuming), or simply remove any excess bit of plastic after the print is complete.  Sometimes, a little hanging plastic may even get lost inside the printed object without producing visible artifacts.

Michele Moramarco
Prusa Research

puterboy, 2023-08-28:
Please address this!!!!

3d-gussner, 2023-08-30:
@puterboy Doesn't this solve the reported issue?
https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.0/Firmware/Marlin_main.cpp#L7582-L7594

GilesBathgate, 2023-08-30:
@3d-gussner I would have thought this line un-retracts the filament to the position before E and L retractions: https://github.com/prusa3d/Prusa-Firmware/blob/v3.13.0/Firmware/Marlin_main.cpp#L3554 

So if you set E and L to 0 there is no issue. But maybe someone wants to keep the retract, but disable the un-retract?

closed 2023-11-06 by github-actions[bot]