prusa3d/Prusa-Firmware #204: Unload filament should preheat automatically

antoneliasson, 2017-09-17:
Because you need to manually preheat the hotend before loading filament due to different materials needing different temperatures, the printer should arguably remember this setting and automatically preheat if needed during filament unloading for a more user friendly experience.

Alomamabobo, 2017-10-04:
Well, preheat the hotend at the right value is not so easy : depend on what filament is loaded :)
My 2 cents.

antoneliasson, 2017-10-04:
Yes, it depends on what filament is loaded -- but you told the printer this value while preheating before loading the filament. My point is that this value could be stored in nonvolatile memory and looked up when the user requests unloading.

antoneliasson, 2017-10-04:
It makes complete sense to assume that the filament was not changed unbeknownst to the printer, since it is not really possible to remove the filament while the printer is powered of and the hotend is cold.

Panayiotis-git, 2017-10-30:
Let's look at an alternative:
Preheat manually, presenting the user a menu of choices like in the Preheat menu. This would be really usefull.

mrjohnc, 2017-11-08:
+1 to this alternative (although I would be happy for either)

belgiancoder, 2018-12-11:
Yep I would put a user friendly wizard instead of A BIG SCARY BEEPING ERROR.
* `Unload Filament`
* `Preheat the nozzle for: -PLA -ABS....` DIAL
* BEEP `You can now remove the filament`
* Detect the absence of filament
* `Load another filament: -YES -NO` DIAL
* If YES, "Insert filament" and autodetect it
* if NO, Nozzle cools down

It is too easy to let the nozzle at high temp after unloading, that leads burned stagnant filament bits clogging nozzle. Also, for someone new to 3d printing it is too easy to get lost in the menu or forget a step, the wizard would prevent that.

leptun, 2019-07-21:
@michalxfanta This is now a feature in the firmware. Should be closed.

closed 2020-09-21 by DRracer