prusa3d/Prusa-Firmware #205: XYZ calibration fails with increased extruder/pinda probe height (eg volcano)

R5fan, 2017-09-18:
It seems XYZ calibration relies on a hardcoded Z height from the top, rather than using the pinda probe. As a result, when the pinda probe is mounted slightly lower to accommodate a ~8mm taller Volcano extruder, XYZ calibration fails and the nozzle scrapes the bed. Some more info in this thread;

http://shop.prusa3d.com/forum/improvements-f14/e3d-volcano-height-adjust-t5896.html

Z calibration does work, using the probe.

madhunm, 2017-09-18:
@R5fan the trick here is to reduce the Z travel. change the max Z to (210 - 8)mm and then check.

3d-gussner, 2017-09-18:
To get both sides perpendicular Prusa asks you during the z- and xyz-cal to move the x-axis until the nemas loose steps, so the firmware 'knows' that both are at ~200mm height. They hardcoded it because it is the simplest way. Otherwise they need to slow down the down movement much earlier not to hit the bed as the height is unknown.

As @madhunm said change the max Z value in the firmware.
Everybody who modifies the hardware has also to deal with modifying the firmware.

PavelSindler, 2017-09-19:
Problem is, that at the beginning of xyz calibration, you don't know where the measuring points are (you are searching for them) and thus you can't be 100% sure that Z endstop will trigger. So I think that Z heigth from the top is correct approach.

etigrel, 2017-10-09:
Can some one be kind enough to assist us on how to update the firmware according to your recommendation?

madhunm, 2017-10-09:
@etigrel Do this. Assemble the print head with the volcano... Do an auto home (Calibration - auto home)

After the homing is completed, move the Z until they stall. Note the value

Open the firmware source in the arduino IDE and change #define Z_MAX_POS 210 in configuration_prusa.h to the value you have noted down.

Compile and flash.

closed 2019-01-29 by inquisitiveKangaroo

nicolai-schlemeier, 2019-10-15:
Could it be an idea to make the machine dimensions in the LCD editable?

3d-gussner, 2019-10-15:
@nicolai-schlemeier you can easy change your height in the source code variant of your printer and compile it yourself.
2nd Comment on closed issues can be overseen as only people opened it and commented get a notification.

Chandler, 2021-04-12:
>Could it be an idea to make the machine dimensions in the LCD editable?

+1 for first class volcano support in the calibration UI

https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-improvements-archive/e3d-volcano-height-adjust/ compiling and reflashing the firmware can be quite daunting for many users