prusa3d/Prusa-Firmware #207: Ability to adjust Lin Advance K factor on LCD

madhunm, 2017-09-18:
Right now, there is no way to change the filament k factor (LIN_ADVANCE_K) when printing. the only way to change it is to change it in the source and recompile.

A feature to change this in the tune menu when printing / settings menu when not printing is really needed.

supermerill, 2017-09-18:
>  the only way to change it is to change it in the source and recompile.

You should also be able to use the gcode "M900 Kxx" in your print script.
As it's surely dependant on the filament characteristics, imo it's a better way to handle it.

GurliGebis, 2017-09-18:
With the K value being calibrated properly and set in the start script, there should be no to change the value while printing

madhunm, 2017-09-18:
@supermerill @GurliGebis I agree... but, when tuning the setting correctly for different filaments, it makes sense to have an adjustment somewhere other than the start code / firmware. 

For example, When switching from PR ABS to eSun ABS; I have found that the eSun is more 'liquid' than PR ABS. I would like to tune that when printing - i can see the changes live rather than end up with a whole load of test parts.

In fact, Marlin has it on the LCD menu as well.

3d-gussner, 2017-09-18:
In Slic3r PE (newest version is 1.37.1) you should add the g-code 'M900 Kxx' in the 'Filament Settings' tab under 'Custom G-code'. Add this custom g-code for all filaments you use.
As mentioned by @supermerill it is a filament characteristics. This g-code will be used when you slice the print.
The K value can be different for filaments, depending on the brand and type, so try to calibrate it with https://github.com/prusa3d/Prusa-Firmware/issues/75#issuecomment-311089517
You can edit the renamed txt g-code temperatures for your needs:
M104 S210 (PLA)
M140 S55 (PLA)
M190 S55 (PLA)
M109 S210 (PLA)

M104 S250 (ABS)
M140 S100 (ABS)
M190 S100 (ABS)
M109 S250 (ABS)

BTW:
M900 K45 is quite good for Prusa PLA
M900 K25 should be a good start for Prusa ABS

Sebastianv650, 2017-09-18:
Instead of using the above linked flat cube I recommend to use the zig zag pattern. More easy to see the differences and you don't end up with a lot of calibration parts.

Also in Marlin I wasn't implementing the LCD option. Somebody else added it later on - as stated above it's not realy useful to have it in the LCD. The risk somebody uses a different filament some day and he will forget to change the value in the LCD because he isn't used to selecting the propper filament inside slicer is too high in my eyes. That's also the reason why K isn't stored in the EEPROM.

madhunm, 2017-09-18:
I guess what I'm asking is this: will there ever be an ability to change the k factor on the fly?

Sebastianv650, 2017-09-18:
Forgot the Link to the pattern gcode: https://github.com/prusa3d/Prusa-Firmware/issues/75#issuecomment-311386115

Sebastianv650, 2017-09-18:
> I guess what I'm asking is this: will there ever be an ability to change the k factor on the fly?

It is, with M900 :-D
But on the LCD, not from my side as long as there is no
- important thing found that makes LCD change useful
- a flood of LCD requests
- or somebody from Prusa team who asks me to do it.

madhunm, 2017-09-18:
:) gotcha.

3d-gussner, 2017-09-18:
@Sebastianv650 You are right I linked the wrong comment :-) 
https://github.com/prusa3d/Prusa-Firmware/issues/75#issuecomment-311386115

closed 2019-01-29 by inquisitiveKangaroo

wavexx, 2020-01-29:
Just resurrecting this for those who care, there is now a define in the latest sources you can enable in ``Configuration_adv.h``: ``LA_LIVE_K``, which adds a tunable to change the K value on the fly.