prusa3d/Prusa-Firmware #210: Odd behaviour after upgrading firmware

joeynuggetz, 2017-09-25:
Been trying to upgrade firmware from 3.0.9 to latest 3.0.12. Firmware flashes ok but if I try to print from SD card, the print doesn't start but instead a Z height adjustment screen is displayed. I was able to print an old GCODE file from Pronterface but if I try and slice something new using latest Prusa Slic3r or S3D and print via Octoprint, the job starts, the extruder heats up and then starts dropping after getting close to target temp and the print never starts. Tried re-flashing a couple times and nothing seems to work. I can flash back to old firmware and then everything is fine.

GurliGebis, 2017-09-25:
Might be some eeprom values that is wrong. Can you try and recalibrate it, and see if that fixes the problem?

joeynuggetz, 2017-09-25:
I did have different E-steps set than default and was thinking it might be related to that. I just flashed back to 3.0.9 and the E-steps appeared to be default. I didn't change/save E-steps this time and just tried flashing to latest once again. Seems like E-step values are different now. 

`SENDING:M503
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z400.00 E144.30
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z12.00 E120.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z500 E10000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1500.00 T1500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.20 E2.50
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P22.79 I1.10 D117.62
echo:PID heatbed settings:
echo:   M304 P126.13 I4.30 D924.76
echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
echo:   M207 S3.00 F2700.00 Z0.00
echo:Recover: S=Extra length (mm) F:Speed (mm/m)
echo:   M208 S0.00 F480.00
echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
echo:   M209 S0
echo:Filament settings: Disabled
`

joeynuggetz, 2017-09-25:
I did some more testing. I tried doing an incremental upgrade from 3.0.9 instead of going straight to 3.0.12. Furthest I can get is 3.0.10 and then octoprint no longer works. This is what I see in the console. This is at version 3.0.11 flashed.  After it hits target temp, you see it temps going back down.
Recv: T:156.10 E:0 B:54.6
Recv: T:155.72 E:0 B:54.7
Recv: T:155.37 E:0 B:54.7
Recv: T:154.97 E:0 B:55.0
Recv: ok
Send: N9 M109 S150*101
Recv: T:154.4 E:0 W:?
Recv: T:154.1 E:0 W:?
Recv: T:153.7 E:0 W:?
Recv: T:153.2 E:0 W:?
Recv: T:152.7 E:0 W:?

I set the extruder to 150 as I was testing and wanted to minimize cool down. It gets to 150 and then it just dies after that point. Temp drops and print never starts. LCD just says "Heating".

I keep seeing differences in E-steps between M503 output after flashing the different firmware files.

GurliGebis, 2017-09-26:
May I suggest doing a factory reset of the printer?
Sounds like something is really confusing the firmware.

joeynuggetz, 2017-09-26:
@GurliGebis I just performed a factory reset and erased all data. Flashed latest Firmware, performed XYZ calibration, set nozzle height (V2Calibration) and the problem persists. Used a brand new S3D profile generated from within the config assistant. Also tried the latest slicer with the same results.

GurliGebis, 2017-09-26:
I think you might want to talk to Prusa support (the chat function on their website).
It sounds like something is wrong with the electronics (would be my guess)

joeynuggetz, 2017-09-26:
I did send them an email and waiting for a reply. I dont think the chat folks are that technical.

joeynuggetz, 2017-09-26:
I decided to try printing directly with Simplify3D attached via USB. That worked no problem. Then I decided to take the Raspberry Pi from my other printer and configure it for the Prusa and suprisingly printing via that Pi/Octo install worked. I recreated a new Pi SD card for my original Pi driving the Prusa and now the jobs appear to be printing. However, I cannot explain why the Z axis move menu is popping up when printing certain GCODE direct from the SD card on the printer. If it wasn't for that I would have tried a new Pi/Octo setup a lot sooner. I'm still not sure everything is 100% resolved but things appear to be for now.

closed 2018-12-28 by inquisitiveKangaroo