prusa3d/Prusa-Firmware #212: Feature Request: Belt driven extrusion compatibility

Quintox303, 2017-10-01:
I recently wrote an article about a new extrusion system that uses a belt to drive an extruder gear:
https://mattshub.com/2017/09/30/belt-driven-extrusion/

The only thing stopping me from making this and testing it out is the fact that if I need to move A distance on the x axis, while extruding B distance, the extruder motor will need to move A+B.. so a firmware change is needed.

In summary, the extruder motor will need to move the same amount as the x motor, plus any extrusion distance (it'll move slightly slower than the x motor when retracting, and slightly faster when extruding).

I would hope that this would be something as simple as extruder_steps=x_steps+extruder_steps in the planner file.. but I'm really not sure. 

@Sebastianv650 I think this would be a really cool thing to implement in the community mod firmware, as just a setting that could be enabled called belt_driven_extruder.

fiveangle, 2017-10-01:
It's a variation on "corexy" as [envisioned by Sven Nielsen about 3yr ago](https://plus.google.com/+SvenEricNielsen/posts/WzexVxrvjK9) (someone else may have come up with similar before but that's when I first heard of it) but as I mentioned in the FB post about this proposal, how does a belt driven E gain anything on a printer where the entire mass of the print bed has to move ?  The moving mass of the X currently is hardly greater (if any ?) than that of the bed, and especially toward the end of a print where the bed also moves the mass off the printed part.

Reducing moving mass of x gantry further wouldn't allow any increase on the acceleration limits for the Y axis so where is the benefit, given the increased complexity and potential for torque-induced flex and belt ossiclations from the orchestrated belt transients ?  Because you can't reduce the mass of the bed+piece, isn't this a non-starter even it it worked "perfectly as expected" ?  Or am I missing something ?

Quintox303, 2017-10-02:
Thanks for that link, I had no idea that it had been thought of before! You're right that the positive effects are mostly pointless on a Prusa machine given the moving bed, but this could also be applied to a regular cartesian with the print head moving on x and y. That's where I think the biggest impact would be. This would just be for testing to see if the extruder belt has any significant impact on the x movement, although I still think it would help. There are moving bed printers like the tevo tarantula and creality cr-10 that use bowden extruders.. It must be for a reason

fiveangle, 2017-10-02:
> There

 are moving bed printers like the tevo tarantula and creality cr-10 that use bowden extruders.. It must be for a reason

Absolutely !  Because Bowden 2 feet of self-supporting teflon tube is a lot cheaper/stronger/easier than securely snaking 4x twisted wires that require support, strain releif, etc, onto the moving gantry.  Plus the gantry can be made more flimsy without impacting the print quality as much.  It has an added benefit that you can use it to support the wires for the HE, so it's a win-win if all you care about is cost.

Quintox303, 2017-10-10:
Fair enough lol. Are you suggesting that the bowden system was only introduced for easy/cheap assembly? What about ultimaker printers?

fiveangle, 2017-10-10:
Ultimaker uses it for the purpose of reducing moving mass of the gantry, in order to improve print speed by allowing higher acceleration over direct-drive (given similar print quality).

Compare the complex but ultra lightweight Ultimaker gantry system to that of the absolute-lowest-cost-movingmass-not-a-concern gantry/bed system of the CR-10 for a prime contrast of opposing design intent.  Ultimaker chose bowden (with all it's faults) for one primary goal: speed.  They didn't choose it because of it's lower overall cost (which it generally is).  Whatever speed benefits one might possibly get with the lower mass of a bowden setup is entirely moot when applied to a printer with a large mass gantry.

Quintox303, 2017-10-12:
I see. Thanks for the explanation :) I know that the benefits of a lower moving mass would be pretty much redundant when you have such a heavy y axis on a prusa printer, but I think that the concept could be adapted to an ultimaker type gantry system. The only reason I'd like to do it on my Prusa is for testing and because it's a much simpler set up that way.

Anyway, regardless of whether it's a worthwhile modification or not, do you think it would be difficult to make those changes to firmware? I had a go at changing the planner.cpp file a little bit with the following

`position[E_AXIS] = lround(e*axis_steps_per_unit[E_AXIS]) + lround(x*axis_steps_per_unit[E_AXIS]);`

But I don't think I have any idea what I'm doing lol. I have some experience with arduino and other languages but not C++.

Would you be able to provide any advice as to how I could do this?