prusa3d/Prusa-Firmware #213: LCD screen not work afterfirmware update

MikeErmagerd, 2017-10-01:
Hi all,

I have been having a heap of trouble with firmware I compiled, after flashing my MK2s the LCD screen becomes garbled looking like

fffffffffffffgggggggggggggggggggvvvvvvvvvvvvvvvvvbbbbbbbbbbbbbbfffffffffffsssssssssssssffffffffffffffgggggggggggggggggfffffffffffffffffgggggggggggggggggggvvvvvvvvvvvvvvvvvbbbbbbbbbbbbbbfffffffffffsssssssssssssffffffffffffffgggggggggggggggggfffffffffffffffffvvvvvvvvvvvvvvvvbbbbbbbbbbbbbbfffffffffffsssssssssssssffffffffffffffgggggggggggggggggfffffffffffffffffgggggggggggggggggggvvvvvvvvvvvvvvvvvbbbbbbbbbbbbbbfffffffffffsssssssssssssffffffffffffffgggggggggggggggggfffffffffffffffff etc

- After resetting the printer the screen becomes alternating rows of white squares and blank rows

- This is my first time so I got a programmer to help me out and we are both getting the same result following the instructions https://github.com/prusa3d/Prusa-Firmware (which are terrible btw)

- There are no compile errors, though two hex files are created...I tested both

Is anyone able to confirm the most recent arduino and prusa driver pack are working on a MK2s?

Any help at all would be appreciated.

Cheers and thanks...Mike

below is the output text from the firmware updater

avrdude.exe: AVR device initialized and ready to accept instructions Reading | ################################################## | 100% -0.00s avrdude.exe: Device signature = 0x1e9801 (probably m2560) avrdude.exe: reading input file "./fw.hex" avrdude.exe: input file ./fw.hex auto detected as Intel Hex avrdude.exe: writing flash (261406 bytes): Writing | ################################################## | 100% 39.83s avrdude.exe: 261406 bytes of flash written avrdude.exe: verifying flash memory against ./fw.hex: avrdude.exe: load data flash data from input file ./fw.hex: avrdude.exe: input file ./fw.hex auto detected as Intel Hex avrdude.exe: input file ./fw.hex contains 261406 bytes avrdude.exe: reading on-chip flash data: Reading | ################################################## | 100% 27.53s avrdude.exe: verifying ... avrdude.exe: verification error, first mismatch at byte 0x3e1b9 0x79 != 0x72 avrdude.exe: verification error; content mismatch avrdude.exe done. Thank you.

3d-gussner, 2017-10-04:
@MikeErmagerd  Sounds bad.

Did you tried to use the Prusa Firmware Uploader which is installed with the drivers http://www.prusa3d.com/drivers/ and re-flashed it back to fw3.0.12 http://prusa3d.com/downloads/firmware/prusa3d_fw_3_0_12.zip? 

Which Arduino IDE version do you use?
I use the newest version 1.8.4 with the Additional Board URL 'https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json' and searching in the Boards Manger for 'RAMBO'. Used in the past the Arduino IDE 1.6.8 with all the changes but don't need it anymore.

When you compile the code and then export the compiled binary make sure that you use the 'Firmware.ino**.with_bootloader**.rambo.hex' file to upload via the Prusa firmware uploader.

**# !!! DON'T USE/UPLOAD THE HEX FILE WIHOUT BOOTLOADER !!!**

Can you still connect via Pronterface (also installed with the Prusa drivers) to the printer via COMx?
If so, try 1st to upload the official firmware 3.0.12.
Otherwise you gonna need to FixMyRambo from Ultimachine. Someone i tried to help also managed to get the miniRambo non responsive and he contacted 1st Prusa and later Ultimachine support.

Hope that helps

stephenwhittle, 2017-10-25:
@3d-gussner Could the compilation/update instructions be updated to reflect this please? I've encountered the same issue after updating to the version without the bootloader. I assumed that the bootloader would be protected, given that we're uploading via FirmwareUpdater which (I presume) is using said bootloader to perform the upgrade...
Following your instructions here, ie using Boards Manager rather than manually copying files over, I was able to reflash without issues, and I have access to the hardware to reprogram the bootloader if need be, but I expect it would be quite scary for people without access to such.

GurliGebis, 2017-10-25:
I don't think it is likely that the documentation will be updated. Based on how long the pull request with the updates has been ignored.
Might be to discourage people from messing with the firmware to begin with (fewer support calls from people breaking stuff)

leptun, 2019-07-21:
@michalxfanta This is very old and doesn't apply anymore. I suppose it could be closed

3d-gussner, 2019-10-08:
@MikeErmagerd  Long time no update here. Can you please close this issue?

MikeErmagerd, 2019-10-08:
.

closed 2019-10-08 by MikeErmagerd

easonjim, 2023-05-12:
#3484