prusa3d/Prusa-Firmware #214: Feature Request: Stop Printing Zone

HTC-Rick, 2017-10-02:
I frequently load up my 3D printer with many models in one print job. When one of the models looses adhesion to the print bed, there currently is one option: kill the print and start over.

What I am proposing is to use the printer control panel to define a stop printing area:
Select the stop printing zone option [implied pause printing].
Move the print head using the control panel to location 1 and store that location.
Move the print head using the control panel to location 2 and store that location.
Calculate the rectangular region of non printing using the two previously stored points.
Make the firmware skip over any commands issued in the "stop printing zone".

This would allow a print to fail in one (or more in the future) area but continue to print the remaining jobs. The desired result would be no more "ramen noodle" prints mixed with good prints.

If this is a difficult or impossible request, I understand and can accept no; however, it would be a nice feature to add to separate the Prusa firmware from the rest of the pack.

dot-bob, 2017-10-03:
FYI: Repetier Server does this and let's you select the print area with your mouse.  You can run it on a PI just like octoprint.

HTC-Rick, 2017-10-03:
I realize that I can spend less than $80 and setup this solution; however, our IT department would not allow a non-Windows 10 anything in the building and would very likely not give me an IP address for the server. Repetier Server would be powerless against SD Card printing as well.

inquisitiveKangaroo, 2018-12-11:
Hi, thank you for your feedback.
That is an interesting idea, however, I am afraid such feature will not be added anytime soon due to the limited capacity of our developers.

closed 2018-12-11 by inquisitiveKangaroo