prusa3d/Prusa-Firmware #219: Feature request: Allow cancelling repetitive beeping on M600

cool-RR, 2017-10-04:
I'm trying to use M600 to make multi-color prints and man, oh man is the repetitive beeping annoying. If the repetitive beeping were to start when I wasn't at home, I wouldn't blame other people in the house for thinking it's the smoke alarm going off, and in any case nobody would be able to sleep until the filament was changed or the printer was thrown out of the window.

Do you think it's possible to add an option in the LCD menu to disable the repetitive beeping on M600? I know I can change a few pieces of code and recompile the firmware but I'd hate having to maintain my own fork of it.

antoneliasson, 2017-10-07:
I think the default behaviour could easily be changed to only beep for a short time, say 30 s, before shutting up. Most other alarms I can think of has timeouts: alarm clocks (both mechanical and digital usually), car alarms and smoke detectors.

The purpose of the beeping should be to grab the attention of a *nearby* user, or give up. If I leave my printer running while I'm away, the first thing I will do when I get home is to check it anyway. It's not like you would forget it. And the buzzer on the MK2 is very loud indeed.

fixing-it, 2017-11-08:
Thumbs up for that. Already thought about desoldering the beeper.

domesticatedviking, 2018-02-06:
Would like the option to disable beep or dramatically reduce it in volume.

3d-gussner, 2018-02-06:
During the M600 filament change the Hotend stays on, and you don't want to leave it hot for a long time as filaments rests in the heatbreak and nozzle start to burn, what can cause clogs if these parts get too big.

It would be nicer if you could make it more silent or even disable it but as the change isn't done within a predefined (safe) time the hotend should turn off, as there is nobody around to do the change.

An Octoprint Plugin which search for M600 gcode height and sends a sms/email/'whatever notification' few layers before would be the best together with firmware option to disable the beeper AND timeout the heater on the hotend.
You could watch TV and get notified via your smartphone/'whatever notfication device'  when user interaction is needed at the printer.

inquisitiveKangaroo, 2018-12-11:
Fixed by adding: _Menu > Settings > Sound: [loud/once/scilent/mute]_ options

closed 2018-12-11 by inquisitiveKangaroo