prusa3d/Prusa-Firmware #221: Resume doesn't work on 3.0.11/3.0.12

filipnavara, 2017-10-04:
This a problem reported on https://shop.prusa3d.com/forum/software-f13/resume-not-working-v3-0-11--t4841.html.

When resuming a print on 3.0.11/12 it is stuck on "heating" progress with the hot end temperature set to 0 degrees.

I can reproduce it on my printer pretty much all the time and I will be happy to give any additional info if necessary. The only thing I can think of that may have caused it is that I sometimes selected the "Pause print" option twice. The second time it was already working on pausing the printer.

Alomamabobo, 2017-10-17:
No problem resuming here on 3.0.12 stock.

filipnavara, 2017-10-17:
Sometimes it works, sometimes not. I haven't figured out the pattern yet. In the case when it doesn't work it shows "heating" with the target temperature of 0 degrees.

filipnavara, 2017-10-30:
As part of some of my printer upgrades, I changed my printer to 3.1.0 firmware from GIT as of June 24. The problem happened again on one of the prints. I paused and immediately resumed the print and the target temperature remained set to 0. I didn't do anything else, so I suspect it could be a timing issue. I managed to recover the print by setting the nozzle temperature back from the menu and as soon as it heated up it resumed where it stopped.

misan, 2018-02-11:
Same problem here, but when resume did not work I paused it again and somehow the printer raised the print head once again and when I finally figured out that preheating may help it started to print on mid air :-( 

I reckon my firmware version is 3.1.1

NateTG, 2018-11-10:
Is it possible that this is caused by pausing again before the nozzle has come back up to temperature?

closed 2018-12-28 by inquisitiveKangaroo