prusa3d/Prusa-Firmware #223: Question: K900 values in v3.1.0-RC1

3d-gussner, 2017-10-05:
@PavelSindler 
**Thanks for the new firmware version and all the effort went to it!**

In https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.1.0-RC1 you mention to use

- PLA: 100 mm/s; M900 K25
- ABS: 100mm/s; M900 K35
- PET: 100mm/s; M900 K45

My tests using https://github.com/prusa3d/Prusa-Firmware/files/1105704/K-Line.Test.txt (gcode) where for Prusa Yellow PLA 'M900 K45' and for Prusa Safty-Orange ABS 'M900 K25'. Check also https://github.com/IsmaelPR1977

I already asked in the https://github.com/prusa3d/Slic3r-settings/issues/18 to add known M900 Kxx values to the filament settings, hope we get soon an update from you and other users.

PavelSindler, 2017-10-06:
Hello @3d-gussner , 
thank you, I will forward these information and questions to Jindrich Benes who is our expert for print settings, materials etc. and he is currently working on it. K settings which I have mentioned in release are not final, we need to do more research, but I wanted to provide some initial value for users to simplify first tests.

GurliGebis, 2017-10-06:
Really, people should print the K-test print (as mentioned above), and use it to calibrate the value for the filament - the same way they use the V2 calibration to adjust Live Z

fiveangle, 2017-10-06:
I agree - for such a new feature where quality of results are so critically dependant on the correct K constant selection, until this is vetted more widely with the community, the defacto standard should be to not recommend specific K but users interested in trying, to calibrate for their specific fillament.

I have found, for example, that 3D Solutech While PLA may require K50, while Hatchbox Black PLA requires only K30.

I would ignore the request to spoon-feed these settings to the community, and instead let us all figure it out in the hive mind until there is more understanding of the valid ranges.  Only with enough collective experimentation may we be able to devise some automatic settings for this, and still get the absolutely optimal results.

With K = 0, the OPRi3 already produces great prints.  Pushing Linear Advance settings that only get it marginally better if not optimized does both PR and the users a diservice :)

fiveangle, 2017-10-06:
Ideally, an easy menu-driven calbiration feature woudl be more useful than autosetting K values.

3d-gussner, 2017-10-06:
I would like getting more pull requests to the https://github.com/prusa3d/Slic3r-settings/pulls. If many requests from the community are published, Prusa could review these and get an average M900 Kxx value to start with for different filaments.

**Not everyone will be able to calibrate the right M900 Kxx value for their filaments, and so I think Prusa should give them a suggestion via the Slic3r Settings before releasing FW 3.1.0 .**

But as said best is everyone calibrates her/his own filaments with the the https://github.com/prusa3d/Prusa-Firmware/files/1105704/K-Line.Test.txt gcode.

How does the printer behaves when M900 Kxx value is not set and Linear Advance Integration is activated?

fiveangle, 2017-10-06:
M900 is always "set" with default of K = 0, which essentially disables it so that it functions as current FW does.

GurliGebis, 2017-10-06:
Why wouldn't everybody be able to calibrate it? It is just a g code file

Sebastianv650, 2017-10-06:
> Why wouldn't everybody be able to calibrate it? It is just a g code file

Sometimes it's hard to see the sweet spot or sweet area. I think we need a quite good recommendation for a start point, but fine tuning has to be done by the user. Similar to estep calibration.

GurliGebis, 2017-10-06:
Indeed, it can be tricky :-)

3d-gussner, 2019-01-08:
Outdated, so  i close it

closed 2019-01-08 by 3d-gussner