prusa3d/Prusa-Firmware #226: Issue: V3.1.0-RC1 slow menus and glitches

jjisnow, 2017-10-05:
Hi, I've noticed a few issues after printing the 1st layer calibration with the pre-release new firmware:

1. 'Pr' in 'Prusa' flashes on and off slowly
2. The menus suddenly slow down dramatically
3. Button presses now shriek instead of give a small click sound - 

The last 2 I believe imply some kind of lagging processing going on.

3d-gussner, 2017-10-06:
Have same issue with the LCD, during the Wizzard the 1st two letters some dots of 'Bed Heating......' message fade off and on slowly.
Also during the Wizzard it switches for a moment to Info screen and then back to the Wizzard between different commands. The Wizzard calibration also runs very very slow sometimes the nozzle stops for a moment.

michalzielanski, 2017-10-06:
I have the same. Even if the printer does not print anything.

PavelSindler, 2017-10-06:
That is interesting. Thank you for reporting this.

WarpZoneNL, 2017-10-07:
- While running setup after factory reset; 4point calibration; it waits a few minutes after calibrate the 4th point!

cristiansavaro, 2017-10-12:
These issues were introduced somewhere between 3.0.12 (no issues) and 3.0.12-7 (lcd menu issues present). I've tested both myself !

PavelSindler, 2017-10-13:
It appears right after turning on the printer, @jjisnow ? Or was it really triggered by running first layer calibration?

PavelSindler, 2017-10-13:
It seems that it is caused by commit https://github.com/prusa3d/Prusa-Firmware/commit/7d2d6c6fdcb296e776c7143cc72f6ca2271aa11 .

jjisnow, 2017-10-13:
@PavelSindler, I only noticed it after printing a first layer calibration, but it may very well have been there from the start.

cristiansavaro, 2017-10-14:
@PavelSindler , it appears random when getting to the menu, but rarely for me at least. Once it appears only reset will resolve it. I hope all the hints can help you in pinpointing the issue and solve it.

PavelSindler, 2017-10-17:
It has been fixed by PR https://github.com/prusa3d/Prusa-Firmware/pull/241.

closed 2017-10-17 by PavelSindler