prusa3d/Prusa-Firmware #227: [BUG] While doing PID under 3.1.0 RC1

bolsoncerrado, 2017-10-08:
"Autotune PID complete!" (note the values!)
https://i.imgur.com/Ero9os4.png

"Invalid M code"
https://i.imgur.com/Jq6GsON.png

M503 delivers OLD PID values :(
https://i.imgur.com/37siWqo.png

SO PID is not storing the new values!!

bolsoncerrado, 2017-10-08:
And just for the record, if you issue the M303 without 0's and spaces, it works fine:

**M301 P19.82I1.64D59.67**
Recv: ok p:19.82 i:1.64 d:59.67 c:1.00
Recv: ok
Send: **M500
Recv: echo:Settings Stored**
Recv: ok
Send: M105
Recv: ok T:35.5 /0.0 B:32.6 /0.0 T0:35.5 /0.0 @:0 B@:0
Send: M503
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E140.00
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z12.00 E120.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X9000 Y9000 Z500 E10000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S1500.00 T1500.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.20 E2.50
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   **M301 P19.82 I1.64 D59.67**

PavelSindler, 2017-10-12:
Hello @bolsoncerrado ,
it seems that this problem can be related to full buffer which is used for reading commands sent over serial.
Is your printer connected to some device which periodically sends something over serial (like M105 gcode)?

bolsoncerrado, 2017-10-12:
Yes Octoprint/Octopi... the one that is going to be supported on MK3? :)

bolsoncerrado, 2017-10-12:
I didnt have any issues calibrating PID manually with other printers via the terminal window on Octoprint tho...

bolsoncerrado, 2017-10-23:
Seems this is as old as....February!?

PavelSindler, 2017-10-24:
I have tried to reproduce this issue many times (with Octoprint and with manually sending gcodes over serial during PID calibration process until buffer was full) and I haven't seen any problem. Also sending M301 gcode from from above (issue description) works well. I asked my collegue who is printing epxert here and is doing some firmware testing now, to focus on it, but he hasn't seen this problem either (with Octoprint). Can you reproduce this issue?

bolsoncerrado, 2017-10-24:
What timeout configurations do you have set on your octoprint? Perhaps it has something to do with that?  As I read other users had this issue and it didnt appear as SOLVED in Github...

PavelSindler, 2017-10-24:
[image]

PavelSindler, 2017-10-24:
Nothing special... You would see timeout in terminal and I timeout issues should be solved now as we started sending "keepalive messages". There are also temperatures sent during PID calibration which prevents timeout issues.

PavelSindler, 2017-10-24:
Did you used precompiled hex file? Which version? Single material Rambo1.3?

bolsoncerrado, 2017-10-24:
3.1.0 RC1  as title states, but cant remember right now if it was on the MMU or the Mk2S :(

PavelSindler, 2017-10-24:
So did you download hex file or did you compile it yourself?

bolsoncerrado, 2017-10-24:
Downloaded.

closed 2019-01-29 by inquisitiveKangaroo