prusa3d/Prusa-Firmware #228: XYZ Calibration Info Menu Option Pauses Print

screwyluie, 2017-10-08:
While you're printing if you go into Support > XYZ cal. details your print will pause until you click out of that screen and then it resumes.

I don't know if it's supposed to do that, but if it is then perhaps it's not something that should be available during a print.

madhunm, 2017-10-09:
@screwyluie when you are printing, that menu shouldnt come at all...

screwyluie, 2017-10-09:
I'm on the latest RC (the one with the linear advance) and it is there, and I was exploring and it paused my print... that'll teach me lol

PavelSindler, 2017-10-12:
@screwyluie, did you print from SD card?

screwyluie, 2017-10-12:
no I was printing from usb

PavelSindler, 2017-10-13:
Xyz details menu pauses printing. It was hidden when printing from SD card, but it was not hidden when printing from USB. It is now hidden also when printing from USB (PR https://github.com/prusa3d/Prusa-Firmware/pull/233 ). This is just hotfix. It would be better to rework xyz cal. details menu to be non-blocking, but it has low priority now.

closed 2017-10-13 by PavelSindler