prusa3d/Prusa-Firmware #232: M110 support is not functional in the 1_75mm_MK2-RAMBo13a-E3Dv6full.hex file

matthova, 2017-10-12:
Just updated my firmware in hopes of adding M110 support. The M110 command does not throw an error, but it also does not set the appropriate line number.

PavelSindler, 2017-10-13:
Hello @matthova ,
can you please provide more information? I would like to replicate the error and I would like to know what actually you did and where M110 was used and what was the result. Thank you.

matthova, 2017-10-13:
Pavel,
Thanks for getting back so quick. Here are the basic steps to reproduce the bug:

1. Update the firmware to 1_75mm_MK2-RAMBo13a-E3Dv6full.hex
2. Connect to the printer
3. Send the following gcode lines:
```
M110; Reset current line to zero
N1 M110 N123*124; Set current line to 123
N124 M114*32; Send line 124
```

Here's the full call and response that I get:
```
TX: M110
RX: ok
TX: N1 M110 N123*124
RX: ok
TX: N124 M114*32
RX: Error:Line Number is not Last Line Number+1, Last Line: 1
RX: Resend: 2
RX: ok
```

matthova, 2017-10-13:
Here's a second test case:

```
TX: M110
RX: ok
TX: N1 M114*38
RX: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0.00 Y:0.00 Z:0.00
RX: ok
TX: N2 M110 N123*127
RX: ok
TX: N124 M114*32
RX: Error:Line Number is not Last Line Number+1, Last Line: 2
RX: Resend: 3
RX: ok
```

matthova, 2017-10-16:
I may have dug too deep. Here's a similar issue I had opened with Smoothieware.
My current plan is to implement checksum identically to prontrface
https://github.com/Smoothieware/Smoothieware/issues/1228
Per @wolfmanj, the change may break support from certain hosts.

thess, 2017-10-16:
Not surprised - given the dumb way a command is parsed, you can only have **1 occurrence** of a letter key (E.g: 'N') per line. My so-called patch tries to cover that. While it works for your examples, it is probably unnecessary.

PR #235 (2017-10-20): Fix parsing of M110 command
Not the best fix but it does work. Fixes #232.
closed 2017-10-20 by PavelSindler