prusa3d/Prusa-Firmware #237: Strange artifacts while printing circles\cylinders FW 3.1.0-RC1

mickstone, 2017-10-15:
[image]

I have been running the beta firmware for just over a week and have found what looks like a bug. Lines seem to print well and at great speed, as do standard rounded corners, but circles do something strange. I was printing some models and noticed really awful looking circles, these were about 25 - 30mm diameter... pictures are attached. Same sliced file with the beta firmware on the left K=40 which is what I worked out looked the best during calibration. I didn't finish printing them because they looked so crappy. On the right, the same gcode but printed after I reflashed FW back to 3.0.12. Nice circles again.

Sebastianv650, 2017-10-15:
Both printed with LIN_ADVANCE or is right with olf FW without it? If it's mixed with and without, try the beta without LIN_ADVANCE to see if it's related to it.

mickstone, 2017-10-15:
Right is old FW, but still had the M900 K40 in the gcode.  I assume the gcode interpreter just errors on it an moves on.  I'm pretty sure I had the same result with 3.1.0-RC1 while having K=0.  I will need to reload the 3.1.0 to test.

bubnikv, 2017-10-15:
Yes, please compare apples to apples first, that means the new firmware
with K=0 set.

Also you may pay attention to "stuttering" of the planner. This happens if
you print faster than the CPU manages to process the G-code lines. The
"stuttering" could be perceived as cyclic slow down and speed up of the
print, the period of this stuttering depends on how far you are from what
the CPU manages: The faster print, the longer is the period.

mickstone, 2017-10-16:
So it looks like I may have been hitting this stuttering issue or I may have had something overheating.  I had just finished a 30 hour print in an enclosure.
Prints done with 3.0.12, and 3.1.0 look so close I can't pick the difference now.  I printed at 40mm\s with K=0 and K=40 with very similar output.  Not sure what will happen when I push the speed but 40 or 60 mm\s is fine for me.

closed 2017-10-16 by mickstone