prusa3d/Prusa-Firmware #239: Feature request: Allow more freedom in the M600 / change filament menu

cool-RR, 2017-10-15:
It happened a couple of times already that I started a multi-color print (using the M600) command, and when the change filament menu came up, my new filament didn't go in correctly. I wanted to push it back up and then try to load it again, but there's no such action in the menu. I had to disassembler the idler wheel to pull the filament out, or lose my print.

I wish that the change filament menu would have more options, like "Move axis" or just "Move extruder axis". This way I could solve my problem without having to disassemble my printer.

3d-gussner, 2017-10-19:
Nice idea, getting an additional option like 'Reload filament' next to 'Filament not loaded' and 'Color not clear' in the LCD menu.
'Reload filament' should be retract filament like in the beginning of the M600 g-code and start again with the procedure.

cool-RR, 2017-10-19:
Thanks. Speaking of which, why does the Prusa extrude 5mm after you click
Yes? This results in about 5cm of filament and I have to very quickly pull
it off with pliers before it gets attached to my print and possibly ruining
it.

3d-gussner, 2017-10-19:
Due to the limited LCD Lines maybe a change of the 'Filament not loaded' function to retract and reload the filament would be easier.

You have to ask Prusa why they use FILAMENTCHANGE_FINALFEED with 50.
Maybe a wipe area in the back of the printer (to prevent the x-axis hits the printed part) could be a solution, a bit like the wipe tower but just on the bottom. Disadvantage is that you loose around 60mm print area in the back.
So i guess cleaning it isn't that bad.

cool-RR, 2017-10-19:
"Due to the limited LCD Lines maybe a change of the 'Filament not loaded'
function to retract and reload the filament would be easier."

Hmm, I would much rather having to scroll down to a separate option of
"Unload filament" than having these two actions bundled as one.

3d-gussner, 2017-10-19:
A scroll down menu is an option.
After thinking what it is about 'Filament not loaded', i think it means 'Filament not loaded correctly' and this should be retraction in case there is some filament in the hotend and redo the loading procedure again.

cool-RR, 2017-10-19:
Well, consider that the user already has a way to get the extruder to spin
forward. We just need a way to spin backward. Please think about a user who
made a print that took 50 hours and now he's changing filament in the
middle of it and he really, really wants to not ruin his print. When you're
in that position, the M600 menu feels so restrictive and frustrating.
That's why I want a straightforward way to roll the extruder backwards,
without automatically rolling it forwards again. These are my two cents, do
with them as you wish.

3d-gussner, 2017-10-19:
I think also that a retract should be implemented in case of having issues with loading filament. We will see what Prusa does with that. As this M600 g-code is used for MK2/s and MK2/sMM they may try to make it as simple as possible for the users AND make sure that the MM also works fine without users retracting manually too far.
As you said we just want to make prints get done, especially long prints!

cool-RR, 2018-04-12:
Hi guys,

Are there any plans to work on this feature? It'd be helpful to me.

Thanks,
Ram.

Chompworks, 2019-06-04:
Also agree this would be a good feature to add to M600. Had a couple of times now where the filament didn't load correctly, but the only option was to try to ram it in further - which isn't ideal.

closed 2023-08-24 by github-actions[bot]