prusa3d/Prusa-Firmware #244: Resume of long pause should notify it's busy till resume is finsihed (e.g. OctoPi)

funkycode, 2017-10-19:
If i want to "long" pause from OctoPi I use after/before script setting to use gcode for Prusa pause.
The problem is when I unpause from OctoPi and code is sent - OctoPi would not wait for extruder to warm up and resumes printing commands.
I this OctoPi should get busy or other identifications till the resume is not done on Prusa and wait for that command to end. (Or is it something that can be done from OctoPi itself?)

PavelSindler, 2017-10-20:
So you are printing over serial (not from SD card) and using our M601 / M602 gcodes inside scripts?
Can you please provide more information?

funkycode, 2017-10-30:
Sorry for late response.
In general in OctoPrint you can add "pre-" and "post-" commands to the pause (before and after). The problem that if you use long pause, after resume it would fire the g-code but did not wait it to finish warming up and would just start printing while extruder is still cold and as far as i remember it did not yet moved to the resume point. 
I'm sure it can be fixed on octoprint too (validate temp and etc). But I think it would be right thing to have commands that would not run new ones in queue till they are finished (except specific one like abort and etc). Not sure if it's possible though...
Once i get home I'll add screenshots and what is configured in OctoPi if it will help

closed 2019-01-29 by inquisitiveKangaroo