prusa3d/Prusa-Firmware #247: Feature request: Add "Max acceleration" to Tune menu

cool-RR, 2017-10-21:
It'll be nice if we could edit the maximum acceleration while a print is in program using the Tune menu.

joeynuggetz, 2017-11-02:
id like this too

leptun, 2019-07-21:
This is not possible since it is changed a lot inside the gcode and it wouldn't stick. Just like with temperatures after a certain layer: If you change it manually and then it is changed in gcode, the gcode value is used

maximlevitsky, 2019-07-29:
Actually this is very possible to do, just the same way as the speed and extrusion flow override is done.
Just make it a percent of the actual value the gcode sets.
In the same way the fan speed could be modified safety from the tune menu, by using a percent,
of the speed that gcode sets.

closed 2023-08-24 by github-actions[bot]