prusa3d/Prusa-Firmware #248: Z-Calibration not working anymore

Boby71, 2017-10-21:
Today, I downloaded the latest files and compiled everything. After flashing the printer, I tried to calibrate Z-axis - and the behaviour was extremely strange:
1) The Z-Axis was pretty slow (I'd suggest 1/3 of the normal speed)
2) There was NO request to go fully up to the upper end of the rods
3) When I tried to move Z up using the "move axis", then it stopped at 201mm height. But there were approx. 5mm missing to the top of the rods where the rattling sound normally appears
4) When I started the Z-calibration from there, the print head digged into the headbed
5) It does the same when I manually (by hand) move the Z-axis fully up to the top
-> it seems like it doesn't care about the upper end or headbed

PavelSindler, 2017-10-22:
Hello @Boby71,
I haven't succeeded in reproducing this error yet. Can you provide more information?
If there were no request to go fully up, what was there instead? How did you move up z-axis using "move axis" menu as there is no posibility to enter move axis menu during z calibration? When you started Z-calibration from 201 mm heigth, what happened?

Boby71, 2017-10-23:
Hi Pavel,
it stopped at 201mm (which was 5mm below the physical end of the rods) when I tried to move it up by “move axis”. The only way to go completely up was to turn off the power, turn the threads manually and the turn it on again. Starting from there, it just slowly came down and digging into the heatbed (which I immediately stopped by pushing the reset button). There was even no try to find calibration points or so.

I stepped back to stock firmware, but will try it later this week to reproduce (compiling and uploading the hex file again)

If it’s still the same, I could send you the compiled file - or even the source code I used as basis.

BR,
Robert

PavelSindler, 2017-10-24:
That would be great.

Boby71, 2017-11-01:
Updated to 3.1.0-RC2 and cannot reproduce this behaviour anymore.  Issue closed.

closed 2017-11-01 by Boby71