prusa3d/Prusa-Firmware #259: MK1 calibration

Roadrock, 2017-10-31:
Hello, I have the Original Prusa MK1 version with 3.0.12 firmware installed, everytime It turn on the 3d printer I get a message about the printer its not calibrated, that i need follow the flow steps from the manual. I  follow the steps from calibration and tests, but when I turn off after a printing a 3d model, and another day I want to print other stuff, the message keeps coming up everytime I turn on the 3d printer.

Any solution?

Thanks

PavelSindler, 2017-11-15:
Did you run V2 calibration? You can try to send G87 gcode over serial or write it to gcode file and print it. Let me know it it helps.

Lauszus, 2018-02-04:
I had the same problem. I connected to the printer via Pronterface and sent G87 and that made the warning disappear.

Before the "Live adjust Z" value is not stored in EEPROM and has to be set manually at each bootup.

Also note that the V2 calibration file is not provided as an stl file in the latest Slic3r release for the MK1 (1.7.7) and can not be found in the calibration menu either.

closed 2018-12-28 by inquisitiveKangaroo